x-station/xstation-issues #199: XStation stopped working after 2 years

leonkiriliuk, 2022-03-03:
A PS1 came back today from a customer I installed the XStation in 2 years ago. The device stopped working. I did a diag on it and discovered the following behavior:

- power LED on
- Access LED flickers and never turns on again
- PS1 boot screen hangs for 10s before the CD player pops up

I’ve verified with multiple good known microSD cards. I verified all the solder points. As a last resort, I swapped the PCB with a good known one and system started working perfectly fine. I swapped the bad PCB back and system not working again. 

I can’t see any physical damage to the PCB. Given the age of the unit, the customer ordered another one for me to swap in. 

Is there anything I can do to help the DEV team figure out what failed? Make sure it doesn’t happen to others?

ramapcsx2, 2022-03-03:
Hey, if you still have the console and xStation, could you try swapping or <very thoroughly> cleaning the flat flex cable contacts?
I don't think that the pcb or parts are bad, but possibly the connector and/or flat flex might.
Try also reseating the cable in a slightly different position, please.

leonkiriliuk, 2022-03-03:
I already tried that with no success.  I've reseated it and cleaned
contacts multiple times.  According to the customer he was playing a game
for a few hours, the screen went black and then the XStation never worked
again.  Using the same system / flex cable, just swapping the PCB tells me
that the flex and console is good.  There's some failure on the actual
XStation device.

leonkiriliuk, 2022-03-03:
> Hm yea, that sounds like that's the case then.
> Is there any way you could send the device to me (in Germany)?
> I'll pay for shipping, of course.

Sure. can you e-mail me your shipping address? My email address is leonki
at
gmail.

ramapcsx2, 2022-03-03:
Could we talk on Discord? :)
I think you're on there, right? Otherwise: https://discord.gg/g56EzEQA

ramapcsx2, 2022-03-03:
The unit will be shipped to me for testing. Thanks, @leonkiriliuk :)

closed 2022-03-03 by ramapcsx2

ramapcsx2, 2022-03-20:
Update: It was one of the connections on the R2 resistor array that has failed. I've messaged you on Discord :)