x-station/xstation-issues #201: Trouble with Japanese Psx

jvivallos, 2022-03-17:
In a japanese 5500 I installed the QSB and it worked. But when I did the "update firmware" the xstation stopped working, I get the "insert a ps1 disc" but only on japanese. I've tried with 3 5500 ps1's already, with 3 new QSB and flex cables, and everytime the same thing happens. Is there a way to know if my xstation is defective in some way?.

ramapcsx2, 2022-03-17:
Hm, that's unusual. I often develop on these 5500 systems, so I'd have noticed if there was a logic problem.
It's unlikely that a working xStation broke right after a firmware update, so maybe it is something related..

Do you have a way to run homebrew on the console(s)?
If you have a way to run Unirom, it will tell the state of the disk system. I can debug the problem then.

jvivallos, 2022-03-18:
I don’t have a way to run homebrew. Can you recommend a way to do it? As far as I know Japanese ps1 s can’t run free psxboot from a memcard.
Also in case I have to, will the ps1 drive work normally?

ramapcsx2, 2022-03-18:
A Gameshark or any of the clones are capable to run Unirom (it's even in the name :) ).
You still need a way to install it though, which means a way to run a CD-R at least.

A console with the xStation QSB installed has some pins lifted and the CPU clock is disabled. This would have to be reversed first, before the drive works again.

ramapcsx2, 2022-03-18:
Ah, and can you try using a fresh SD card, with just the "00xstation" folder and "loader.bin" inside it?

ramapcsx2, 2022-03-18:
And another update: An NTSC-J console can run freepsxboot :)

jvivallos, 2022-03-22:
> Ah, and can you try using a fresh SD card, with just the "00xstation" folder and "loader.bin" inside it?

I just tried this and worked. Seems like the update kills it somehow. Do you know what could be causing it?

ramapcsx2, 2022-03-23:
Okay, so we know there is some kind of problem around the sd card setup.
Maybe the other card is formatted badly? Maybe it doesn't have the /00xstation/loader.bin layout?