x-station/xstation-issues #202: Xstation SCPH5000 JAP but Motherboard LATE PU-8 : Shows PS1 logo and then black screen.

miguelk5, 2022-03-18:
Hello,

I did an Xstation install on a PS1 NTSC-JAP that have SCPH5000 but the motherboard is a LATE PU-8: 1-658-467-21

It have also the 3 component output just like the SCPH1002 aka audiophile.

SCPH Label:
[image]

Motherboard:
[image]

So I followed the LATE-PU-8 guide, but some screenshots of the guide were quite different:

Filter and resistor:  Guide
[image]

My PS1:
[image]

IO016 : Guide:
[image]

My PS1:
[image]

I've checked the install:

-I've checked all the QSB joints with digital microscope and they seems to be fine, but I resoldered it and added more tin just in case.
-The chip CXD2510Q pins and they are okey too.
-The filter and resistors removed have no rests of tin.
-I've checked with another flex cable.
-I've tested with another Xstation unit and another working SD card.

[image]

I'm out of ideas, can be that there is not specifict guide for this model ?

ramapcsx2, 2022-03-18:
Hey,
the behaviour alone is not sounding good. You get a black screen if the system saw valid data, and additionally it recognized the license symbols.
The system wants to start, but then something goes wrong.
It sounds like chip damage on the motherboard to me..

Any way to run Unirom?

miguelk5, 2022-03-18:
> Hey, the behaviour alone is not sounding good. You get a black screen if the system saw valid data, and additionally it recognized the license symbols. The system wants to start, but then something goes wrong. It sounds like chip damage on the motherboard to me..
> 
> Any way to run Unirom?

Hello thanks for your quick response. It can be a configuration problem since I had tested with a memcard with games inside different folders, example USA / SHOTING / House Of The Death / hoft.cue. And I recall that my brother have a 1002 and he can't setup correctly the config for folders, so I pickup a new SD card and formated, placed the last firmware and without games it loaded the Xstation menu ! Now I'm adding 2 games without folders, I'm going to re-test it and let you know

miguelk5, 2022-03-18:
It worked, seems like when the Xstation s configured to read sub folders from an Ps1 SCPH5500 it doesn't boot if connected to a PU-18 LATE, so you need put a clean SD card and the games without sub folders. Maybe there is a problem with this feature on the Late PU 18 models.