x-station/xstation-issues #203: Tomb Raider [SLUS-00152CE] - Shows all .bin files under games list

lluked, 2022-03-19:
Hi,

As the title, Tomb Raider [SLUS-00152CE] Shows all .bin files under the games list, it is the only game I have noticed do this. I have checked the Cuesheet and Track 01 is listed as `MODE2/2352` and all others are down as `AUDIO`, I have even re-downloaded the Cuesheet from [redump.org](http://redump.org/disc/1471/) and refreshed the games list. I'm not sure if is because the Game has so many .bin files (57) it has met an unexpected condition, I haven't seen another game with this many so far, or there is something else wrong with the Cuesheet.

Kind Regards
Luke Davies

ramapcsx2, 2022-03-19:
Hey,
if this happening on a recent firmware and loader? The latest is 1.4.5 currently.
There are filters that should hide the extra .bin files, the .cue is not involved in that.

lluked, 2022-03-19:
> Hey,
> 
> if this happening on a recent firmware and loader? The latest is 1.4.5 currently.
> 
> There are filters that should hide the extra .bin files, the .cue is not involved in that.

Hi,

Thanks for the quick response. Yes the latest firmware 1.4.5. The files all get picked up, the dump is verified with redump.org hashes, I'm not too sure what else I could have done wrong.

Thanks,
Luke

lluked, 2022-03-19:
> Hi,
>
> Thanks for the quick response. Yes the latest firmware 1.4.5. The files all get picked up, the dump is verified with redump.org hashes, I'm not too sure what else I could have done wrong. [...]

+ Latest latest loader aswell

ramapcsx2, 2022-03-19:
Can you share a screenshot of how the filenames look like on the sd card?

lluked, 2022-03-19:
@ramapcsx2 Thanks, you've just found the problem, for some reason every track is saved as (Track 01) (`Track Number`).bin instead of (Track `Track Number`).bin. I feel stupid now for not noticing before, the names didn't even line up with the redump Cuesheet.

closed 2022-03-19 by lluked

ramapcsx2, 2022-03-20:
Ah right, the filters aren't very smart :)
I chose processing speed over more thorough checking, as it starts to matter a lot with more files.