x-station/xstation-issues #205: Will not get passed the insert disk screen (PU18)

ahinkle117, 2022-03-28:
Hello,

I recently installed my x-station into my SCPH-5501 NTSC-U Playstation, but I can never get passed the "insert Playstation Disk" screen when I turn it on. I verified the continuity between the pins on the QSB and the test points on the board. I also made sure that the proper pins were lifted on IC 701. I've also tried again with a different SD card, but nothing got me passed the insert disk screen.

The following is the top of my board:
[image]

and the following is the bottom of my board:
[image]

I'm at a loss at this point as what could be wrong with my x-station, and I would appreciate any other suggestions on what to test.

ramapcsx2, 2022-03-28:
Well, I can say that the error condition means that signals on the QSB towards the left side (where the hook is) are not working.
But also, the rest of the installation doesn't look like there'll only be this one problem.
The installation needs to be redone. It'd be great if you could send it to an installer :p

ahinkle117, 2022-03-28:
I was hoping to avoid the cost of an installer, so that's unfortunate. Thanks for the help though.

ramapcsx2, 2022-03-28:
Well, I can't predict whether you'll be able to "reflow" the QSB. Repairwork that doesn't fix the problems is making the situation worse.
An installer already has to assume that there are shorts and such, making their work more difficult.
I'd rather outright recommend the installer option here.

ramapcsx2, 2022-03-28:
Worst area here. I think there are shorts, and the original circuit might also not work anymore. It looks like the small SMD parts moved..
https://i.imgur.com/Ns647Ss.png

ahinkle117, 2022-03-28:
I guess I mangles it more than I thought I did. I'll look into which recommended installer will be cheapest and have the quickest turn around time. Again, thank you for the support.

closed 2022-03-28 by ahinkle117

ramapcsx2, 2022-03-28:
Good luck!
I also found this area on the upper side, though it would not affect your errors. It's still something that has to be cleaned.
I think some moisture made it into the console once.. Something like that.
https://i.imgur.com/b6g0W7r.jpeg