x-station/xstation-issues #206: Black screen no logo nothing

cryptblood1986, 2022-03-30:
Sony cxd2510Q correct pins lifted. no shorts. no bridges.
correct filter cap removed and the resistor beside it. correct 3 resistors removed under the board.
all qsb points checked. no shorts. no bridges. checked the board ic pins continuity ok

There appears to be missing components on the left side. The unit came like that.

Unit will turn on without logo or sound. Only idle blue light.

[image]
[image]
[image]
![275637207_267489555584215_9171367308544696491_n](https://user-imag
[image]
es.githubusercontent.com/83506761/160784544-74be19d4-ba8c-48d5-86a2-4850052acc05.png)
[image]
[image]
[image]
[](url)

ramapcsx2, 2022-03-30:
If the console came with parts missing, those parts are not important. I use a PU-8-41 (NTSC-J, right? 3000 series or so?) as well.
No video means most likely the APLL clock isn't reaching the CPU. From the pictures, the soldering looks great.
The ESP32 chip should power up, and it will start sending a clock signal down the APLL line. This clock signal has to reach the CPU clock input.

cryptblood1986, 2022-03-30:
Is there another way to check apll clock issue?

ramapcsx2, 2022-03-30:
An oscilloscope is the go to tool for debugging hardware.

cryptblood1986, 2022-03-30:
[image]

[image]

[image]

The unit gives a split second green light flashing before going to blue.
Regulator gives 0.0v 3.3v and 7v.

ramapcsx2, 2022-03-30:
See the single small resistor on the xStation PCB near the flat flex connector? This is the APLL clock line. You can measure whether it reaches all points.

ramapcsx2, 2022-03-30:
The xStation doesn't even have to have any SD card inserted, and you'd still get a picture. (It would then boot to the BIOS shell menu.)
I'm a bit at a loss on what to recommend. It seems like you did everything right, and it should work.
Was the console working before? IE the cables, upscaler, etc, are known good?