x-station/xstation-issues #207: Possible to solder 467-11 pu-8 early via jumper?

cryptblood1986, 2022-03-31:
[image]
Is it possible?
I can micro solder well.
If there's a pinout, it'd be super useful.

ramapcsx2, 2022-03-31:
Yes, it's possible. You would need error checked and precise documentation, which I sadly don't have in one place.
The matching QSBs are in production (have been for a while), and should be available at all new vendors, as well as at Castlemania soon!

ramapcsx2, 2022-03-31:
Here is the annotated board, but I'm not 100% sure about every point. Ideally wait for the QSBs, which will work.
[image]