x-station/xstation-issues #209: Not rebooting after update

brendan1495, 2022-03-31:
Hi, could someone tell me the reset point on the 100x QSB I have looked at the Mini FAQ and that just shows the reset point on the 550x QSB also is there any documentation that explains what each point is, I know they are numbered but as a noob I have no idea what the numbers correspond to.

ramapcsx2, 2022-03-31:
This pic is from the EARLY revsion, but the reset point is the same: "21" nearby the square Mechacon IC:
[image]

cryptblood1986, 2022-03-31:
I'm having the same problem. Where do I connect the 21_RST to the 100x qsb?

cryptblood1986, 2022-03-31:
[image]
[image]
Realized my reset pin is burnt. Where can I jumper it to?

brendan1495, 2022-03-31:
Thats what I am trying to work out no 17 on the above diagram I think is the reset, but where can I connect it to suppose I could try the pin on the chip but is there an easier point, tried this point but that doesn't seem to work either as my board isn't exactly the same

[image]

ramapcsx2, 2022-03-31:
Check with multimeter: One of the power supply connector pins should be the reset line.

brendan1495, 2022-03-31:
> Check with multimeter: One of the power supply connector pins should be the reset line.

Hi thanks for the reply, but I dont really know how the qsb works. If I connect the pad on the qsb to the power supply reset line will that work, I am pretty sure I know which one it is but dont want to fry the board.

brendan1495, 2022-03-31:
Hi all, thanks for everybody who gave me advice, it actually turned out nothing was wrong with the install reset, I was using a replacement PSU 12v, changed it to a standard ps1 PSU and all good, its like there was not enough power to reboot, strange, but all OK now.

cryptblood1986, 2022-03-31:
Having problem, after connecting the reset pin to pin 17 of the qsb my unit worked for a while then now not working....

If I remove the reset pin it is still the same not working....