x-station/xstation-issues #210: Black screen again after xstation install pu-8 model 21 when connected reset pin

cryptblood1986, 2022-04-01:
[image]
[image]
![20220331_204052]
[image]
Unit was working before the reset pin was not connected.
Board is a Japanese pu8 21. I had exact issue with another Japanese pu8 47. 

Now both units give black screen even with out the xstation and all parts and mods restored to stock.....

Feels like an incompatibility issue with Japanese units.

ramapcsx2, 2022-04-01:
It's certainly a soldering mistake.
- I've tested nearly all possible variations of boards, and most only differ in some software details or the reported regions
- restoring all and it still fails means there is a fault

cryptblood1986, 2022-04-01:
Would it be safe to test the unit with only the qsb attached without the xstation connected? 
It would still show Sony logo right?

ramapcsx2, 2022-04-01:
No, it can't work without the CPU clock. So you need to have the xStation attached, so it delivers the APLL clock.
In addition, the QSB bridges some points on the mainboard (by design), which would cause problems if you just restored the original CPU clock.
You can either run the xStation modified console with an xStation attached, or completely undo the mod (remove the QSB as well).

ramapcsx2, 2022-04-01:
Where are you located?
Would it be reasonable to send the console to me (in Germany)?

cryptblood1986, 2022-04-02:
I'm located in philippines. Is there a way for me to discuss with you privately?

ramapcsx2, 2022-04-02:
Okay, sending to me won't be a good idea then :p 
But sure, I'm always on Discord:
https://discord.gg/zDAMbjjA

Annunaki, 2022-05-15:
Solder point 17 is not even soldered, looks like it's been missed.