x-station/xstation-issues #211: xStation won’t boot - black screen!

MJ4Tech, 2022-04-01:
The PlayStation turns on and the tv detects video but there is no sound or picture.

Where should i look first?

Pictures of installation attached 

Your advise is highly appreciated. 

Thank you!

[image]
[image]
[image]

ramapcsx2, 2022-04-01:
Well, look for shorts, check power arrives at xStation, check each QSB solder point is making good contact, etc.

MJ4Tech, 2022-04-01:
[image]

ramapcsx2, 2022-04-01:
It'll also help if you insert the flat flex the right way around :)
https://raw.githubusercontent.com/x-station/xstation-releases/main/install_pics/PU-18/pu18_14.png

MJ4Tech, 2022-04-01:
> It'll also help if you insert the flat flex the right way around :)
> 
> https://raw.githubusercontent.com/x-station/xstation-releases/main/install_pics/PU-18/pu18_14.png

And i was running in circles, now I can't wait for my shift to finish so i can get home and try flipping it and see if that would fix it! Thanks a lot

MJ4Tech, 2022-04-02:
> And i was running in circles, now I can't wait for my shift to finish so i can get home and try flipping it and see if that would fix it! Thanks a lot

It's working! Thanks a lot

[image]

ramapcsx2, 2022-04-02:
Sweet, glad this was a quick fix. Enjoy :)

closed 2022-04-02 by ramapcsx2