x-station/xstation-issues #213: NTSC game videos not playing on PAL PlayStation

brendan1495, 2022-04-02:
Hi All,

When I try to play Some US games  on my pal Playstation after the ps logo shuts down I get a blank screen, I can sometimes hear the music and if I mess with the controller I can get into the game and play it, its as if the videos wont play on my tv even though I have changed the setting of videos from auto to ntsc. I have my playstation running through a OSSC to hdmi, is this a problem with my TV do you think? all pal games run fine.

ramapcsx2, 2022-04-02:
Yes, that's definitely a video problem.
I can only speculate, but maybe the OSSC to TV chain doesn't like the slightly off-spec field rate.
This would be down to the TV then.
It could also be that the OSSC may have to be set to enable a force 60Hz option (if it has that).

brendan1495, 2022-04-02:
Will have a look at the OSSC, thanks for the reply

Sseerii, 2022-09-30:
Greetings. Installed xstation in console 1001 ntsc. I use it to connect RGB Scart to a CRT TV. NTSC images work fine. But PAL images don't work. In most cases, this is a dark screen, or just sound without a picture. I think the xstation is installed correctly, because that's the only problem. If you switch the picture mode to PAL in the menu in the menu, then there will be a dark screen and the firmware has to be updated again. I know that xstation should remove regional restrictions, so this worries me. Why is that? If I buy a PAL set-top box, do I understand correctly that the problem will be only in black and white, which will be solved by RGB scart? Why doesn't the reverse order work for me? Thank you.

Potolococo, 2022-12-21:
> Greetings. Installed xstation in console 1001 ntsc. I use it to connect RGB Scart to a CRT TV. NTSC games work fine. But PAL games don't work. In most cases, this is a dark screen, or just sound without a picture. I think the xstation is installed correctly, because that's the only problem. If you switch the picture mode to PAL in the menu in the menu, then there will be a dark screen and the firmware has to be updated again. I know that xstation should remove regional restrictions, so this worries me. Why is that? If I buy a PAL console, do I understand correctly that the problem will be only in black and white, which will be solved by RGB scart? Why doesn't the reverse order work for me? Thank you.

Hello, I have a question. Were you able to solve the problem? The exact same thing happens to me.

brendan1495, 2022-12-21:
I never did

ramapcsx2, 2022-12-21:
Hey,
as I mentioned before, there can be a few different reasons that NTSC consoles won't work with PAL software / PAL video mode.
The TV could not like the signal (same with upscalers), especially since the PAL rate will be off spec when using an NTSC console.
Then there are some consoles that have a modification (from the factory) that simply doesn't clock the GPU in PAL modes. A modification is required for those.
Lastly, the TV / scaler problem has variations, such as that it might work with Composite Video, but not RGB, and vice-versa.

In all cases, the xStation does nothing for or against this. It is neutral with this issue :p

Sseerii, 2022-12-22:
Hello! Yea, we need a installation PS1 dual frequency oscillator.
It`s only for 1001 console. 5501 is worked without DFO, use only RGB.
My theme: https://github.com/x-station/xstation-issues/pull/247

Potolococo, 2022-12-22:
> 

> Hello! Yea, we need a installation PS1 dual frequency oscillator. It`s only for 1001 console. 5501 is worked without DFO, use only RGB. My theme: #247

Sorry if I ask a lot, but I would like to be able to solve this issue.
Do you have an image of how the modification should be in an American plate SCPH-1001 PU-8?
I was reading in forums to change a resistor, but it did not work for me, it continues to be black when running a PAL game.
I leave you an image of how the work turned out.

[https://i.ibb.co/y6bkGZ8/aaaaaaaaaaaaa.jpg](url)

ramapcsx2, 2022-12-22:
The modification with the swapped resistor is correct for these NTSC-U console models. This resistor swap enables the PAL GPU clock, so that part is now fixed.
If any one of you look at this section, and you have a resistor in the marked old location, you also need to swap it, as shown.

Next, @Potolococo, please try using different video cables or TVs / scalers. If it doesn't work still, the TV/scaler could just not like the "almost PAL" frequencies.