x-station/xstation-issues #214: XStation Overheating and Plastic Stand Pins

cristobalflorian, 2022-04-06:
Hi, and sorry for my bad english...

I have a Xstation but already lost the plastic pins to settle down the board, so I'm using a plastic bag where it came to avoid the short. 

Mi first issue is, where I get those pins, I bought another QSB but is just the QSB and flex cable, no pins
The second one is that XStation is overheating at the first 30 minutes of usage... I dont know if this problem is related to the first one, or my installation have some flaws, but it works great, but when its hot any game freezes and ps| cant boot.. 
I changed the PSU thinking it was the problem, but still overheating... 

Thanks

cristobalflorian, 2022-04-07:
> Enviado el: jueves, 7 de abril de 2022 07:20
>
>
>
> Hey, let's see:
>
> *	I don't know where the plastic pins are from, sorry! Maybe you can find one, which is enough to stabilize it on a PU-18.
> *	Heat builds up near the xStation because there is no ventilation at this place in the console. This has been well tested though, and the area may get warm, but never get close to overheating. There is about 0.5W of heat energy generated on the xStation, which is not enough for concern.
> Since you do get problems though, I suggest you try cleaning the flat flex connector thoroughly (it might make bad contact).
> Also clean the SD card pins.
> For better ventilation, make sure the console doesn't stand on a carpet.
> *	The font size in the loader are a compromise between readability and amount of information. Have you tried the other color scheme or adding the background?

Thanks for your reply Robert… 

 

I see, maybe changing the caps on the psu can solve it, since I exchange it from another model 1001, I have an early PU-8

About the plastic pins that comes with the box I thiking to print them in a 3D printer using the size of the holes on the board as reference… 

 

The XStation is really awesome, it’s more than a piece of hardware, it’s about preserving my CD collectoin, and the most important thing, the superb nostalgia feeling when I play all my childhood games… Thanks for helping me…

ramapcsx2, 2022-04-08:
I'm glad you like it :)

Heat should not be an issue at all, even in an EARLY PU-8 with the worst power supply.
The worst (highest) temperature I measured was on such a machine, with the LDO regulator reaching some 63C, if I remember correctly.
That is hot enough that you can't keep a finger on it, but it's still ~20-30C below the limit for it.
Other parts are always well below the LDO temperature.

LDO: http://www.advanced-monolithic.com/pdf/ds1117.pdf
125C junction temp, assuming junction is 40C hotter than the surface

I think the problem is something else, and I hope you can find it!

Annunaki, 2022-05-14:
I'm having the same issue with the overheating on a late PU-8 board, after a while the games and menu crash and also the X-Station is extremely hot to touch.

Annunaki, 2022-05-15:
You can find the plastic PCB stand-offs on Amazon :)

cherryuk, 2022-05-20:
Sorry for jumping onto someone else's post, but I'm having heat problems during the hot weather. A desk fan a meter away could keep the PS1 & X-station cool for as long as it needs, but without it and in a room of 20ºc+ I could only get 1.5hrs of gameplay before sound/loading issues occur. Would a heatsink on the x-station chip or underfan like a laptop fan or PICO PSU swap help with the thermals at all. I only ask this question as I have heat issues in the Summer, and it's only late May at the moment, so not too hot. I have a 1st batch of LaserBear mount and understood with an 8bitmods query that the extender ribbon was the cause of sound glitches. This has been removed and so it's just the mount with standoffs. Any help with cooling would be great.

ramapcsx2, 2022-05-20:
Well, it should not be possible, but I won't doubt your experience. 
Maybe there is some kind of fault in the machine that develops with heat.

The area where the xStation sits has zero ventilation, if you think about it.
All the machines' heat rises and sits right below the CD lid area.
So if you put the console on its side, that should help with that. Can you give that a go?

Ah and no, there is no part on the xStation that gets too hot and could be cooled with a heatsink :p

ramapcsx2, 2022-05-20:
If you've noticed the problems in a particular game, is it only in this game?
And if so, may I know which it is? :)

cherryuk, 2022-05-20:
Thanks I'll try those tips, it's probably due to being upstairs on a warmer evening, in the cooler months there were no issues. So i'll try a few things first downstairs and then the console on it's side. It's a PAL SCPH-5552 model, this was checked Oct last year by 8bitmods to all the soldier points, all good and checked the flows.

Games played are - Robbit Mon Dieu (JPN), music starts cutting out and then the loading stops on to the next screen and then on to a black screen. Weird thing that happened was that I activated the desk fan opened the lid, closed it and then the game carried on loading to the next screen and with the fan on continued to work perfectly and the top of the Playstation was cool to the touch.

ramapcsx2, 2022-05-20:
It might be a bad lid switch sensor.

BK2K, 2022-05-23:
I would just like to add my experiences with heat and electronics.
I have many computers in a commercial environment and over the years have learned that just because something is within its rated tolerance its still best to keep it as cool as possible. I would have video cards die after only a year even though they never got "too hot".  After increasing ventilation and not allowing them to maintain their own fan speed but to be max all the time I am going on almost 10 years now no death.
Bottom line is extreme heating and then cooling when powered off over and over can kill electronics in time.
Moral is no matter the specs the cooler you can keep something and more heatsinks you can put on the better.

ramapcsx2, 2022-05-23:
Yes, that's the general rule.
I can't repeat it enough though, xStation generates 0.5W by itself. 
It gets warmest at the LDO location, which has good ground flood to dissipate that heat.
Heat expansion stresses apply to the sd card slot at best, but that area is out of any hot zone, and way below any limits.

It's a phenomenon where all the console heat accumulates in the area, and people assume their issues may be due to heat, but it's not.

ramapcsx2, 2022-05-23:
Ah well, but if anecdotal evidence helps any, I was working in 35C room temperature last summer, daily, and overheating parts never were an issue :)

cherryuk, 2022-05-25:
Putting the PS1 on its side solves the heating up issue. The heat comes out of the vent on the side, the heat that used to be on the top near power totally vanished. Thanks for the tip and might look into Todd’s Retro Frogs 3D vert stand.

ramapcsx2, 2022-05-25:
Hehe, alright, if that works for you :p 
I would still look into the lid switch for the CD tray. The fact that it was able to resume gameplay after opening/closing the lid means that nothing actually froze. It might be that the heat makes it so that the lid switch briefly disengages. The mechanical tolerances are too little with this design.