x-station/xstation-issues #215: Lifted a pad?

Smallmodd, 2022-04-20:
Issue with ps motherboard so I'm going to mod a different one. Had to remove x station and I think I removed a pad too...what can I do?
[image]

ramapcsx2, 2022-04-20:
Find out where the damaged via connects to on the flat flex connector, then solder a wire from there to the PSX solder point :)

Smallmodd, 2022-04-20:
> Find out where the damaged via connects to on the flat flex connector, then solder a wire from there to the PSX solder point :)

Thanks for your reply! How would I go about that, is there a diagram of the x station I could follow? I circled the one I think I lifted 
[image]

Smallmodd, 2022-04-20:
> Find out where the damaged via connects to on the flat flex connector, then solder a wire from there to the PSX solder point :)

- I'm mainly having trouble figuring out if it's on the 4th or 5th. Going crosseyed 😂

ramapcsx2, 2022-04-20:
It's the 5th pin on the flat flex connector (counted from the left):
https://user-images.githubusercontent.com/6278722/112033652-f08d7380-8b3d-11eb-9124-131df4cb646f.png

Smallmodd, 2022-04-21:
Amazing, me and my eyes thank you!!
[image]

Smallmodd, 2022-04-21:
> It's the 5th pin on the flat flex connector (counted from the left): https://user-images.githubusercontent.com/6278722/112033652-f08d7380-8b3d-11eb-9124-131df4cb646f.png

Amazing! Me and my eyes thank you 😁
[image]

ramapcsx2, 2022-04-21:
Good luck with the install :)
If you have a multimeter, the "beep test" is a great tool to ping out which connection goes where :)

Smallmodd, 2022-04-24:
Well, for some reason it's no longer working so I bit the bullet and ordered a new QSD :') Tried it on both mobos and it was working before - I have bad luck when introducing wire into QSD installs