x-station/xstation-issues #216: No Boot at all - late PU-8 model

M4k4d4m, 2022-04-21:
Hello all !
Bruno, from France. Hope I'll get some nice help here :)
To be honest I'm completely lost as I tried everything possible (I guess) to solve the issue.

The issue is simple, the xstation won't boot.
I am having the Sony logo and then the console goes to the memory card / music player screen, nothing else.
I can see the "access" and "power" LEDs get lit when I power on the console.

(Note that I only tried it while the parts were not reassemble yet but it would not make any difference I guess).

I strictly followed the installation instructions and the steps provided in the help section too.

What I've done to try to solve the issue : 
- Reflow all the solder joints > DONE
- Continuity check from all the PINS of the xstation to the components on the board > OK
- Formated twice the SD card and downloaded a fresh file from the website > DONE
- Continuity check on the lid switch > OK

Really... Please someone help me <3

Bruno

ramapcsx2, 2022-04-21:
Hey,
please take some detailed pictures of the install, so I can get an idea of the situation :)

M4k4d4m, 2022-04-23:
[image]

M4k4d4m, 2022-04-23:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

M4k4d4m, 2022-04-23:
[image]

M4k4d4m, 2022-04-23:
Hi and thanks for the quick reply. 
please find here the detailed pictures of the installation. Hope the pics are ok. 
Tell me if you need more specific ones 😉👍

M4k4d4m, 2022-04-23:
[image]

ramapcsx2, 2022-04-23:
Great, I don't see any obvious problems at first hand, but some solder points look worse than others. Those are most suspect.

What happens when you enter the CD player and hit Start?
You want want to see a CD playing, but near no sound.
The CD timer should slowly go up.

M4k4d4m, 2022-04-23:
Finally the issue got worse… I plugged the controller / memory card part and I immediately lost the video signal. Now I can’t get nothing on the screen, no boot from the mother board at all. Just a wavy kind of signal 😢
[image]

M4k4d4m, 2022-04-23:
[image]

ramapcsx2, 2022-04-23:
Hm, I can't make out what that CRT picture is showing.
The power loss when plugging something into the controller port most likely means that a fuse blew. That occasionally happens, especially with 3rd party controllers and such (but sometimes it just happens).
Try to find the blown fuse with the multimeter beep test.

M4k4d4m, 2022-04-24:
> Hm, I can't make out what that CRT picture is showing.
> The power loss when plugging something into the controller port most likely means that a fuse blew. That occasionally happens, especially with [...]

I’ll get to that fuse later when I come back from work. I have a 3rd party controller yes which I use for tests on consoles. 
then I’ll come back to you after doing the steps you provided. Thanks a lot for the precious help

M4k4d4m, 2022-05-02:
Hello !
I’ll try out the few steps u gave me. 
just to be sure I tried an another SD card and I soldered the XStation to an another identical board I had and the issue is the same. 
what do you think ?

ramapcsx2, 2022-05-02:
Can you do the test I asked for?
What happens when you enter the CD player and hit Start?
You want want to see a CD playing, but near no sound.
The CD timer should slowly go up.

M4k4d4m, 2022-05-02:
I finally did the steps u provided and it does not behave as mentioned, nothing happens in the CD page. Can’t press play, no CD reading. 
[image]
[image]

ramapcsx2, 2022-05-02:
Okay, the basic stuff works, but either data doesn't get recognized, or the drive is not unlocked.
Do you have a cheat module for the expansion port?
If so, can you flash Unirom to it and see what that does when trying to start a game?
https://github.com/JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1

M4k4d4m, 2022-05-04:
What do you mean by cheat module ?

ramapcsx2, 2022-05-04:
Ah well, I mean those old cheat modules that can run custom code on the machine. 
If you have to ask, you probably can't run Unirom then...
( https://i.imgur.com/vAfQjpQ.png )

Can you hop on Discord, so we can talk about you sending me your console? (If you want to go that route :p )
https://discord.gg/heNST4Ku

M4k4d4m, 2022-05-04:
Nerves used one of this modules for sure but I bought the console to a guy so maybe he did some stuff with it but I don’t think so really. So yes I can come so we talk about it directly on discord. 
thing is The time zone isn’t the same so I might catch you in the evening when I get back from work, or now. It’s midnight here. 
I have no issue sending the PSX. I more concerned about shipping and border fees which will apply 😅

M4k4d4m, 2022-05-13:
Up !
Did you get my previous message. Can you tell me when and which channel should we use to talk if needed ?
Thank you 😊

ramapcsx2, 2022-05-15:
Hey,
is the link still valid? I'm always on there (okay, wasn't this weekend :p) :)

M4k4d4m, 2022-05-15:
Ok thanks :) 
What’s your name on discord ?
Mine is M4k4d4m #9312
Add me and send me a PM if you want