x-station/xstation-issues #217: Boots to blue screen

Sushko123, 2022-05-02:
Hi I did the mod and it just boots up to the blue screen. With the CD lid closed it freezes and unfreeze  I can go to the CD menu and play one track with no sound but it plays. The xstation lights are blue on continuous. The green light comes on after the Playstation sound and go off when blue screen comes up. If you leave it on the green light will go on and off. I tested all test points from motherboard to the ribbon cable and they are all good. I looked over the lifted pins and they are fine. I even tested continuity on the pins next to the lifted pins. I have the 2 files in the 00xstation folder and it made a configuration file. I also have one game. I don't have a memory card installed. 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ramapcsx2, 2022-05-02:
Hey,
please multimeter beep test for shorts to ground on all QSB solder points. Use the connection diagram for this. Only the points marked with a star (*) should have ground:
https://user-images.githubusercontent.com/6278722/112033652-f08d7380-8b3d-11eb-9124-131df4cb646f.png

If that checks out, please fix that missing connection shown in one of your pics, where it clearly has no solder :p

The problem is definitely with the installation, as you get some action in the CD player.

Annunaki, 2022-05-02:
What happened to pin 44 on CXD2545Q? Is it missing or is it a camera/light thing?

closed 2022-05-02 by Sushko123

ramapcsx2, 2022-05-02:
Good work! :)