x-station/xstation-issues #218: BlueRetro (ESP32 module)

Z3R0XZ, 2022-05-09:
Hello, I'm trying to make Blue Retro work with Xstation but seems something interferes with the ESP32 module and don't work properly.
I do some tests, the module works on an unmodified PS1, PS2 and with the Xstation with the tray open, but with Xstation enabled (CD tray closed) the Blue Retro module start behaving strange, some games show unsupported controller others connect and disconnect the controller (Blue Retro) very rapidly some others press random buttons, then works fine for some time.
My first thought was a power issue, so I use the 7v line insted 3v to power the Blue Retro and the same thing, then I use external power supply for the module and still the same behavior.
I did all the tests with Blue Retro installed directly on the PS1 controller board and using a controller cord is same behavior, I'm clueless.
This is the project, is rellay cool:
https://github.com/darthcloud/BlueRetro

ramapcsx2, 2022-05-09:
Hm, that's certainly weird. Can you clarify where these dropouts happen?
Is this a thing in the loader only, or does it also happen in, say, the BIOS CD player?

Z3R0XZ, 2022-05-09:
> Hm, that's certainly weird. Can you clarify where these dropouts happen? Is this a thing in the loader only, or does it also happen in, say, the BIOS CD player?

Yes, the module works normally, as soon I close the CD tray the cursor disappears randomly and pushes random buttons, this behavior is similar when I connect an external and internal Blue Retro module in the same controller port.

ramapcsx2, 2022-05-09:
We had some talk about erroneous inputs with BlueRetro on Discord the other day. There are some known issues with some PU-18(?) motherboards.
But I haven't heard it connected to the xStation yet. Maybe it's radio interference?

Z3R0XZ, 2022-05-09:
> We had some talk about erroneous inputs with BlueRetro on Discord the other day. There are some known issues with some PU-18(?) motherboards. But I haven't heard it connected to the xStation yet. Maybe it's radio interference?

I forgot to mention but my motherboard is a PU-18