x-station/xstation-issues #219: Menu Freezing

Annunaki, 2022-05-14:
Hi,

X-Station arrived, so I promptly soldered it into place alongside the X-Reset, no problems played all day yesterday completed a few titles however I have noticed when it's left on the Menu (the game list) with the falling snow after an extended period of time it will freeze and become unresponsive.

Cycling the power will fix it, until the next time it's left on the menu for an extended period of time, in which it eventually freezes again.

Is this a known issue?

Cheers

Annunaki, 2022-05-15:
OK... Somehow the jump wire from point 12 broke, I did a continuity test on all points and found the jump wire failed continuity intermittently when moved about, Oddly, the break was right in the centre of the jump wire.

Replaced the stranded jump wire with a solid core wire, and it seems to have worked, The PlayStation has been on the X-Station Menu for about 12 hours now and not crashed or locked up.

I will leave this thread open for several more hours, at which point if all remains OK, I will then close it as resolved.

ramapcsx2, 2022-05-15:
Oki, great you've found the problem :)

Annunaki, 2022-05-15:
> 

Thanks, It's a fantastic bit of kit! Is there a Discord or discussion Forum for X-Station?

ramapcsx2, 2022-05-15:
We took up residence on the Classic Gaming Discord :)
https://discord.gg/R7rjQWNN

Annunaki, 2022-05-15:
Closed this now, This issue was to do with the jumper wire from point 12 (PU-8 setup) having an intermittent connection, which was causing the freezing of the X-Station Menu and crashing of games.

closed 2022-05-15 by Annunaki