x-station/xstation-issues #220: xstation black screen after update

MrFilz, 2022-05-15:
I updated the firmware on my xstation today, now it gives me a black screen with an purple vertical line on the left side of the picture. It worked perfect before the update... Can i roll back or try to update again, or is the xstation now broken?

Annunaki, 2022-05-15:
> I updated the firmware on my xstation today, now it gives me a black screen with an purple vertical line on the left side of the picture. It worked perfect before the update... Can i roll back or try to update again, or is the xstation now broken?

I'm not sure how you can roll back without menu access, I would for the sake of it test all points for continuity to rule out any possible installation issues that might have coincidentally popped up.

MrFilz, 2022-05-15:
i will do that the next days. I wanted to install new caps and a new psu anyway...

ramapcsx2, 2022-05-15:
Hey,
the update mechanism is such that it won't commit the update if anything is bad. The problem may have developed from a bad ESP32 flash chip though, which I've had happen once before.
If it worked perfectly, then after the update not at all anymore (green LED not blinking, only the blue one is on), then please contact the seller of the xStation for a replacement. (It might be the flash chip again..)

MrFilz, 2022-05-15:
Good to know, i will test and look for the blue and green led lights! Thanks a lot