x-station/xstation-issues #221: X-Station and Xplorer Carts

Annunaki, 2022-05-24:
Getting a strange issue using the PU-8 X-Station ODE with an Explorer Cheat Cart on Metal Gear Solid, The intro FMV and Audio will "skip/jump" and a CD icon will flash on and off in sync with the CD icon, When eventually in the game's main start screen the music is not skipping but is playing at a faster than normal BPM or whatever.

The game will load into a playable state just fine, but any cut scenes etc will be jumpy/skippy.

Annunaki, 2022-05-24:
If I replace the Xplorer Cart with my Action Replay Cart or use no Cart, then the issue goes away, I will do some more testing by flashing the Explorer Cart with other Xplorer ROMs via Unirom and then post back here.

Had a quick test of some other games, but at the moment it seems limited to MGS in my findings.

ramapcsx2, 2022-05-25:
We had a chat on Discord, right?
The problem is that some Xplorer/Xploder install a handler that listens to a button combination in-game.
Whatever the handler does, it breaks many games. Only disabling the function (if possible) can fix this.