x-station/xstation-issues #222: Please, Help to install PU-7

maniaco007, 2022-05-29:
guys, I bought the Xstation, my board is the PU-7.
How to install?

ramapcsx2, 2022-05-30:
Hey, we don't have a quick solder board (QSB) for the PU-7 series. 
The earliest boards for which a QSB is available is the EARLY PU-8.

It is possible to solder in the xStation manually, but I would not recommend it. In addition to the difficult work, the support of these revisions is badly understood. For example, the loader might misbehave.

I recommend you look for a later board, ideally PU-18.

maniaco007, 2022-05-31:
Unfortunately I will have to wait a long time now to be able to use Xstation.
I'm from Brazil, to be able to buy the xstation I sold a kidney....

ramapcsx2, 2022-05-31:
I understand. 
You will have to go through all the issues here that talk about either EARLY PU-8 or PU-7, and piece together the required steps. This issue has the most information:
https://github.com/x-station/xstation-issues/issues/52

Also reference with the LATE PU-8 installation debug image, as some points are similar:
https://camo.githubusercontent.com/1c8ab0e280c04d7aa869175ce70adb2e36547936168dd837fb953dab8d50664e/68747470733a2f2f692e696d6775722e636f6d2f675448673363612e6a706567

maniaco007, 2022-06-16:
I bought another console.
I did the installation and I'm loving Xstation!
Consider creating something like this for the Sega Saturn!

Thank you for creating this wonder!

ramapcsx2, 2022-06-16:
Glad you like it! :)
Saturn is well served with Fenrir, which is very similar to xStation :)
https://www.fenrir-ode.fr/

closed 2022-07-06 by maniaco007

Tyson1911, 2022-12-31:
Is there anything new in the installation of a PU7 board?  I'm trying to do it myself but fail with no result