x-station/xstation-issues #223: Start Problem with Resident Evil 3 Nemesis und Dino Crisis 1

Manuell64, 2022-05-30:
Hello, everyone

Have a problem with 2 games on the X-STATION. Resident Evil 3 Nemesis and Dino Crisis 1 . The two games start normally up to the info screen with the message This Game contains scenes of explicit violence and gore. And then it kind of freezes and the game doesn't carry out any further actions. Have reinstalled the games (Resident Evil 3 Nemesis & Dino Crisis 1) several times. And today the MicroSD card was formatted again with an extra tool (https://www.sdcard.org/downloads/formatter/ .

I really hope you can help me further.

ramapcsx2, 2022-05-30:
Hey,
is this on a PAL console? And which region are the games?
If anything is PAL, it is especially important to have updated to the latest firmware, which is 1.4.5 as of this time.

Manuell64, 2022-05-31:
Yes, it is a Pal console. And both games are also pal versions. I played the last update on the console. All other games on the console and also Pal versions all run without problems. Only Resident Evil 3 Nemesis and Dino Crisis 1 have this problem. And the games just mentioned do not show the green start screen either

ramapcsx2, 2022-05-31:
Okay, can you tell me which specific release Dino Crisis is? There are releases for PAL UK, Germany, etc.
Ideally the SLES number would be great :)

Manuell64, 2022-05-31:
Dino Crisis SLES no. 02209
German version
Resident Evil 3 Nemesis and Dino Crisis 1 work on a Youtuber (Spielwelt) on his X-STATION and why it doesn't work for me is a mystery to me ;(

Outlook für Android<https://aka.ms/ghei36> herunterladen

ramapcsx2, 2022-05-31:
Okay, I can test that version. It's the .de one :)

ramapcsx2, 2022-05-31:
The game works fine here.
Maybe your game has been modified or incorrectly dumped. 
It's possible that 1.4.5 doesn't have all the required patches yet, so to test that, please wait a little while. The new update is ready any day now, fingers crossed.

Manuell64, 2022-05-31:
Thank you dear Robert for your help.

Outlook für Android<https://aka.ms/ghei36> herunterladen

Manuell64, 2022-05-31:
Dino Crisis 1 (Europe) Backup was from CoolRom. Start the game normally until the screen freezes. Without the Piracy green screen. (also Resident Evil 3 Nemesis) then I repeated a CD-Rom (German) with INGBurn for Dino Crisis 1 and got exactly the same error as just described above. I keep my fingers crossed and hope that the next firmware update will do better.

Outlook für Android<https://aka.ms/ghei36> herunterladen

ramapcsx2, 2022-05-31:
Yes to both.
It's like a conventional release: You can go back and forward to any other version, as you like.

ramapcsx2, 2022-06-06:
The full 1.6.1 release is out now. Please try that with your games.
It should work just fine, but if not, first thing I would check was the game rips.

Manuell64, 2022-06-07:
Hi, thanks for the information . But how do I install the update on the SD card? Do I have to delete the old version from the SD card and then install the new one on it?

Outlook für Android<https://aka.ms/ghei36> herunterladen

ramapcsx2, 2022-06-07:
There are 2 files in the update .zip. Extract them both onto your sd card, into the 00xstation folder (replacing old files, if they exist).
Then start the console to the loader, go to the menu and select update :)

Manuell64, 2022-06-07:
Thank you for your prompt reply . That's how I did it. Console shows Firmware: 1.6.1. Unfortunately Dino Crisis 1 still doesn't start :(

Outlook für Android<https://aka.ms/ghei36> herunterladen

Manuell64, 2022-06-07:
Same with Resident Evil 3 Nemesis

Outlook für Android<https://aka.ms/ghei36> herunterladen

ramapcsx2, 2022-06-07:
I tried the German version of Dino, and it had no issues. I can also only speculate that there's a problem with the game images..

Manuell64, 2022-06-08:
> Sowohl RE3 als auch DC1 PAL funktionieren einwandfrei. Stellen Sie sicher, dass Sie REDUMP-Bilder verwenden und nicht von einer Rom-Site.
> 
> Beide PAL-Titel haben Libcrypt-Schutz, aber X-Station unterstützt das.
> 
> Wenn es jedoch bei REDUMP-Dateien auftritt, könnte dies auch an separaten Sprach-SKUs liegen, dh in Ihrem Fall an der deutschen Version, da dies darauf hindeuten würde, da ein Hängenbleiben des Gewalt- und Gore-Bildschirms der libcrypt-Schutz ausgelöst wird.
> 
> Auch hier unterstützt die X-Station-Firmware Libcrypt, aber möglicherweise könnte es ein Problem mit der deutschen RE3-SBI-Datei geben.
> 
> Versuchen Sie, die mehrsprachige PAL-Version (Redump) zu testen. Wenn das funktioniert, dann ist das Problem die Ger-only-Version libcrypt.

Thanks for the hint.
Where can I get the redump version?
Have already tried the German version of Resident Evil 3 Nemesis with the current firmware version. It just freezes on the Violence and Gore screen. And as already mentioned, no Piracy notice appears in either game. It appears in every game that I installed on the X-STATION.
I'm really sad about that. And what does that mean with the game pictures.

Greetings and thanks again for your support

Manuell64, 2022-06-08:
Hello, when I ripped Dino Crisis 1, game images were an extra file. However, I did not install them on the X-station

Outlook für Android<https://aka.ms/ghei36> herunterladen

ramapcsx2, 2022-06-09:
Sorry, when you ask for where to get games, I can only tell you to get them from your disks, using Imgburn :)

What do you mean with " It appears in every game"? Earlier you said that all other games are working, right?

ramapcsx2, 2022-06-09:
I just tried Resident Evil 3 (German) with Imgburn. It creates a .bin and a .cue. Copy both to the sd card and run > works fine.
This was on a PAL console as well, so all checks passed fine.

Manuell64, 2022-06-11:
I would like to send you a video. But Outlook keeps telling you to compress the message. Which I do.
Only it doesn't work. Too bad.

Outlook für Android<https://aka.ms/ghei36> herunterladen

ramapcsx2, 2022-06-11:
Oh, I believe you that it's not working for you :)
Since other games work, that means there is only one thing that can be wrong: The game rips are bad.

Manuell64, 2022-06-12:
> Oh, ich glaube dir, dass es bei dir nicht funktioniert :) Da andere Spiele funktionieren, bedeutet das, dass es nur eine Sache gibt, die falsch sein kann: Die Spiel-Rips sind schlecht.

Good evening ramapcsx2
The recording clearly shows that everything started normally. The rip can't be bad as Dino Crisis was previously pulled from the internet like all other games. Only when the European version didn't work did I rip the German version. Just as I had ripped Oddworld Abe's Exoddus it worked correctly. As said before, both Dino Crisis and Resident Evil 3 Nemesis do not show the privacy screen on startup. This means that they freeze in the Violence notice or is not running

Manuell64, 2022-06-12:
Here the records. Tried to upload a video but it doesn't work again or I'm too stupid for it :) [image]
[image]
[image]
[image]
[image]

Manuell64, 2022-06-12:
Oddworld Abe's Exoddus, which I also ripped using the same program as Dino Crisis, is the only skeleton game that displays the privacy screen and runs smoothly ;)
[image]
[image]
[image]

ramapcsx2, 2022-06-14:
Hm, I dunno. I made sure to do a fresh rip of RE3 from my disk, using Imgburn. 
Imgburn is a simple program and there are no bad settings that could mess it up.
The required patches to get these extra protected games working are all included in the firmware. 
The user doesn't have to do anything more.

It should work ><

retrofank, 2022-06-14:
yep, they all work fine on my two machines and a friends.   If there were issues, others would have reported it by now, since RE2 and DC1 are popular titles, so it appears to be a isolated case for you. 

1. Could be the source files. (try looking on Archive.org) redump.

2. PC HDD, try different PC

3. SD Card/Adapter.  Try a different SD card (reformat it correctly)

Manuell64, 2022-10-11:
Hello I ripped Oddworld Abe's Exoddus with Imgburn as well as Dino Crisis 1 and no problems. I'm just thinking that the SD card might not work. It's almost brimming with storage. Could that be because of this?

Outlook für Android<https://aka.ms/ghei36> herunterladen

Manuell64, 2022-10-11:
Hello I have now ripped Dino Crisis again with games picture and installed it on the X-STATION. It starts the Sony and Playstation logo. And the latter remains frozen. What else can I do? :(

Outlook für Android<https://aka.ms/ghei36> herunterladen

Manuell64, 2022-10-11:
Based on the photos I show in this order that the games do not want to start.

Outlook für Android<https://aka.ms/ghei36> herunterladen

ramapcsx2, 2022-10-11:
It's been a while. Please remind me what you've tried so far :)
It might be the SD card indeed. Can you try any other card, just to see whether it works?

Manuell64, 2022-10-12:
Hi

Yes, I just used a new SD card.

Same problem . I find it very sad. Ironically, the two top games

Outlook für Android<https://aka.ms/ghei36> herunterladen

ramapcsx2, 2022-10-12:
Sure, I can understand the frustration. But we know these games are supposed to work. The xStation has no trouble with making them work.
To find out why it doesn't work for you, I need to know whether anything about your console or the game rips or the sd card, etc, is unusual, that might cause it to fail.

Manuell64, 2022-10-13:
It's just how all games work. It also says that Resident Evil 3 Nemesis and Dino Crisis 1 have the same copy protection. And exactly these 2 games only work up to the violent screen. It's really just annoying. Also have a Nintendo64 with an Everdrive64. And just any Rom goes without problems. And I got the Playstation games Rom's from the same platform. From here I think it may be due to the copy protection of the games.
That really annoys me, because I was looking forward to these games the most, since they are not available on the Nintendo64 :)

Outlook für Android<https://aka.ms/ghei36> herunterladen

retrofank, 2022-10-13:
Again, those two games work perfectly fine (yes there is copy protection i.e Libcrypt but that is already taken care of) and is not a X-station issue, otherwise there would be many countless bug reports for them, since they are popular flagship titles.   

Try a different console maybe?

Manuell64, 2022-10-13:
I only have this Playstation.
Getting another one would be far too expensive. This console has already cost over 350 €.

Maybe you can email me the roms. To test.
If you say that your ROM's work perfectly on the X-STATION.

Outlook für Android<https://aka.ms/ghei36> herunterladen

retrofank, 2022-10-13:
We have already advised a few months ago to use Redump in this thread.   

Search on Archive.org for Redump PS1.  The PAL multi language and the seperate Deutsch version is there.

FloppySoft, 2022-10-13:
You could compare your rom with the redump one, e.g. via hashes. If you ripped your legit copy on your own, chances are high there are issues with the rom. If you got one from the internet, chances are high, too, to receive bad copies. Another game, Omega Boost, is a perfect example of a game with a bad copy being distributed online everywhere.

retrofank, 2022-10-13:
> I only have this Playstation. Getting another one would be far too expensive. This console has already cost over 350 €. Maybe you can email me the roms. To test. If you say that your ROM's work perfectly on the X-STATION. Outlook für Android<https://aka.ms/ghei36> herunterladen
> […](#)

[image]

retrofank, 2022-10-13:
[image]
[image]

Manuell64, 2022-10-13:
What exactly should I do on Archiv.com? Don't get it exactly?

Outlook für Android<https://aka.ms/ghei36> herunterladen

Manuell64, 2022-10-13:
> Click view files after doing the search for Playstation Redump. It will list them like my picture above ^^. Then click on one of them to download it.

Hi

I really thank you
Thank you for writing to me again. I have followed your instructions . Resident Evil 3 Nemesis and Dino Crisis 1 run as intended. Thanks alot.

I can finally play these 2 games.

Outlook für Android<https://aka.ms/ghei36> herunterladen

ramapcsx2, 2022-10-13:
Thanks for helping, everyone. Seems this is solved :)

closed 2022-10-13 by ramapcsx2