x-station/xstation-issues #224: Does not boot past white screen - No loader boot - No Memory Card/CD player screen

pdiggitydogg, 2022-06-05:
Looking for ideas.
I've installed an Xstation onto a late rev PU-8 1-658-467-22 that now refuses to load past the white bios screen. 
***It does not load the memory card screen.***

There's no distortion in the image and the melody plays completely, so I'm not sure the bios is corrupt (which I guess is a thing on pu8)
It (mostly) loaded game discs prior to install.

I've reflowed the QSB and checked continuity to all points, from the board's respective vias/ic legs to the ribbon connector on the QSB, so the install seems good.
I've reflowed the bios and dram as well as all other legs on IC309 (cxd2510q)
Lid switch works properly.

I've done quite a few of these but I've never had one fail in this way. 
The xstation with respective ribbon is confirmed to be working with the same SD card on 2 other consoles (both pu18s).
I've reflashed the SD card several times, with and without game titles, different loader revisions and again after swapping it into a PU18. Config file is written to the card.

Does anyone have thoughts on what the issue could be?

[image]
[image]
[image]

ramapcsx2, 2022-06-05:
Does the Memory Card screen appear if you start the console with the lid open?
If it does not, you have a problem with the machine elsewhere of xStation install points. You mentioned reflowing chips.

pdiggitydogg, 2022-06-05:
No, the memory screen does not appear, regardless of the lid switch position.
It worked fine before the install, which is the strange thing... I attempted reflowing ic's suggested elsewhere for bios issues, since it seemed to be that, but could it also be a handoff issue between the bios and loading the xstation or the card screen?

ramapcsx2, 2022-06-05:
That's unlikely. It would be more likely that something shorted with the installation. Look for solder balls between IC pins and such.
Also thoroughly check your reflows, as those are now suspect.

pdiggitydogg, 2022-06-05:
Appreciated. Reflows happened after there was no boot the first time after rechecking the installation, I do not suspect them.
I'll close as it's believed to be the console not the kit

closed 2022-06-05 by pdiggitydogg

ramapcsx2, 2022-06-05:
Hmm, still wish you good luck, maybe you can find the problem!
It's definitely not getting a response from one of the chips, or the response is bad, so the shell code hangs..