x-station/xstation-issues #228: oscillator jitter

nvnv-sunny, 2022-06-16:
I found a serious problem when upgrading the PS audio DAC, the XS output clock accuracy is not good, even after 4 divisions to obtain the 16.9344MHz clock will produce a dynamic offset, the result is that it will increase the jitter of the digital signal, deteriorating the sound. Is it possible to use the host clock input to the XS instead of the clock generated by the XS.Maybe the SPU's master clock (33.8688MHz) must be synchronized with the CPU clock, otherwise it will affect the game sound.Because  i want to transform PS1 into a HIFI digital Player, so i must to find a solution to the problem.

nvnv-sunny, 2022-06-17:
Thanks for your help. I'm designing a new audio DAC circuit. To get better results, the DAC require high precision and low jitter clock signal, but the XS output clock doesn't seem to be high precision, so I'm looking for a solution.

nvnv-sunny, 2022-06-21:
I should have found the solution, thanks again.

ramapcsx2, 2022-06-21:
Well, good luck!