x-station/xstation-issues #229: Koudelka Spanish version : When kill final boss no credits appear, it go directly to the initial screen - Stuck when change room sometimes

miguelk5, 2022-06-17:
In the Spanish version of Kouldelka (I don't know if the rest versions too)

When I load more than one disc with full or fast load sometimes the game just stuck when changing from one room to another, and you will see just a black screen.

Then at the end of the game once I defeat the final boss without matter what ending is (the game have many endings) the game just bypass the credits and go to the initial screen, but I saw that the game have credit screen: https://youtu.be/o0Jz-cCcx4c?t=1482

retrofank, 2022-06-19:
On the latest firmware? 

Could be the SD card?  fragmented or a slow access one?

Try a different one (formatted correctly) with just that game only to test with again.

ramapcsx2, 2022-06-20:
Hey, what is the quickest way to get to such a hang?
Can you share your memory card save file somehow?

ramapcsx2, 2022-06-21:
Okay, I still have this and some old saves. Maybe I can reproduce it :p 
[image]

ramapcsx2, 2022-06-21:
I played the game for an hour and a bit, did a couple cutscenes and beat the plant boss. I didn't find any issues or odd things in the logs.

retrofank, 2022-06-23:
Had a quick look at the English version and that seemed okay. Didn't get stuck or had delays when entering/exiting rooms like the OP mentioned with the Spanish version.