x-station/xstation-issues #230: Feature Request: Custom themes, or more themes in general

Ritaku02, 2022-06-23:
The winter theme from the previous update has me wanting more themes in general, or a way to make my own themes using images. The person who programs the FXPAK/SD2SNES firmware has an online tool where you can put in the menu file from the firmware, choose colors, and upload an image for the top banner, to create a custom menu. Something like that for the Xstation would be nice. Failing that, just having more themes to choose from in the menu would also be nice.

ramapcsx2, 2022-06-24:
Upcoming :)

Ritaku02, 2022-06-25:
Cool, looking forward to it! :D

maniaco007, 2022-06-25:
Cool, looking forward to it! :D  (2)

ramapcsx2, 2022-06-25:
There'll be a loader update (surely this weekend) that lets you pick between classic and winter theme, nothing more yet, sorry :p 
For full theming as seen on flash carts and such, we want to open source the loader. Too many different inputs are required to make this good, it's a community thing.

Ritaku02, 2022-06-26:
> There'll be a loader update (surely this weekend) that lets you pick between classic and winter theme, nothing more yet, sorry :p For full theming as seen on flash carts and such, we want to open source the loader. Too many different inputs are required to make this good, it's a community thing.

That's fine, I'm just happy that it's something that's being looked at and being considered, even if ultimately it'll have to become a community effort.

maniaco007, 2022-06-30:
The theme swap was perfect!
Thanks for implementing this feature!

Ritaku02, 2022-06-30:
I'll echo that, thanks a bunch!