x-station/xstation-issues #231: Point Blank 2, unreliable boot with 1.6.1 after updating from 1.4.5

highhike, 2022-06-26:
I've got two SCPH-5501 PSX with xstations I installed.  Both work great, and one PSX is at a dedicated light gun setup.  I recently flashed the upgrade for 1.6.1

Point Blank 2 has issues getting past the intro movie now.  Hitting start during the intro movie to skip will often lead to a black screen.  Letting the movie complete after the initial load will usually get the title screen, but after some time the movie starts a 2nd time and after it ends it won't load the title, just a black screen.  I had the same results on both xstations with 1.6.1 installed.  I tried different downloads for the Point Blank 2 disc image with the same results.

Long story short, I flashed one of the xstations back to 1.4.5 and Point Blank 2 is loading reliably again.  I let it do the movie to title screen back to movie loop for some time without issue.

ramapcsx2, 2022-06-26:
Oh, a core bug? Thanks for reporting, I'll take a look :)

highhike, 2022-06-26:
wow, thanks for the fast response!  A lightgun connected or not doesn't seem to matter.  I have a Guncon in port 1 and controller on port two for menu navigation on the one PSX, the other Playstation at the main setup has just a controller connected.  Skipping the movie with the controller or a gun button press would often crash it with 1.6.1

Full boot does not appear to make a difference.  I just started Point Blank 2 on the console that has 1.6.1 installed.  The movie started and I hit start to skip, and it made it to the title.  I let the title sit for a while and the movie started again.  After the movie completed the game crashed to a black screen.  I don't think this is a SD card problem as this is the same SD card, I'm just moving it between systems.  I put the card back in the light gun PSX with 1.4.5 (fast boot) and it's running a movie to title to movie loop without crashing.

ramapcsx2, 2022-06-26:
Can you still try a different SD card? There's a chance your card is marginal, and different error handling in 1.6.1 now fails in this game.
It's on loop on my machine for an hour or so now, no problems :)

highhike, 2022-06-26:
I found an older 16GB SanDisk full size SD I had laying around.  Fresh formatted it FAT32 and put Point Blank 2 on it.  The other card I've been getting the crashes with is a much larger microSD SanDisk in exFAT.  

This second card with Point Blank 2 will still crash on 1.6.1.  I was able to skip to title, movie start, skip to title a few times but eventually I got a black screen.  I just did a fresh load of the game (fast), and it failed after the second playthrough of the movie leaving the controller undisturbed.  I put this 16GB card in the 1.4.5 machine and it seems to be working fine.

ramapcsx2, 2022-06-26:
Okay, I guess it's unlikely that 2 cards fail in the same way..
Are there any more details you know about your console? Does it have any other mods installed?

ramapcsx2, 2022-06-26:
Update:
I have one more report that the game fails, so it's definitely something going on.
I'll have to find out what's different.

highhike, 2022-06-26:
Now that is some amazing tech support!  I have absolutely no problem leaving the light gun PSX at 1.4.5 : )  I'll flash it again when the next update drops.  Have a great Sunday!