x-station/xstation-issues #232: ESP32 fails to start after firmware update, I suspect booltoader area is damaged, what should I do

ribenyamazun123, 2022-06-26:
ESP32 fails to start after firmware update, I suspect booltoader area is damaged, what should I do

ramapcsx2, 2022-06-26:
Hey,
the ESP32 updates either fully, or it rejects the update. It is brick-proof when using the normal update method.
What happens when you start the console now? Any LED activity?

subjectivelyobjective, 2022-07-05:
I've had this issue as well. It just stays on the white Sony logy screen now. There is a solid blue Power LED, and the Access LED blinks once at boot and then never again.

ramapcsx2, 2022-07-06:
@subjectivelyobjective
Hey, your problem is something else. 
Try starting the console with the CD lid open, and go to the CD player. Close the lid and play the disk.
Does the timer run up, looking normal?

subjectivelyobjective, 2022-07-06:
The timer doesn't run up at all unfortunately. There's no green Access LED either when I attempt to play either.

ramapcsx2, 2022-07-06:
( If it's a problem, no worry. But could you record a small video of the power up, showing the green LED and ideally the TV? )

ramapcsx2, 2022-07-12:
@subjectivelyobjective's issue got solved, but @ribenyamazun123 didn't get back to me. Is this still relevant?