x-station/xstation-issues #234: Game will not switch on multidisc mode

mikeriosjr, 2022-07-18:
Ps1 550x

Game won’t switch when I open lid and close lid. 

I create a playlist on the loader. When game prompts to switch disc, I open and close lid and game with either prompt enter incorrect disc or it will go to black screen.

The game acknowledges that the lid opens up during the first game so that works but nothing after works correctly.

I checked the txt file and both games I select  show up in the multigame list.

I’m thinking I may not have something soldered correctly but I’m not sure at this point

ramapcsx2, 2022-07-18:
Hey, could you put 2 CDDA using games in a queue and run the BIOS CD player?
Open the lid and hit reset to go to the BIOS menu, then select the CD player. Close the lid and check whether the CD can change at all.

I agree with you: The lid open/close detection is working. So at this point, I don't have a good idea yet.

mikeriosjr, 2022-07-18:
So I just tried what you said and no luck. 

The audio disc doesn't change over to the second audio in the playlist. 

If there is a contact on the QSB i need to check please let me know.

ramapcsx2, 2022-07-19:
Okay, I think the lid switch signal isn't connected to the xStation, so it only works on the console side, and games detect it.
If you have a multimeter, please check that this signal arrives at the xStation's ESP32 module:
https://i.imgur.com/fSfIEVi.png

It should read around 100 Ohm between that point and "IO35" on the module, shown here:
https://4.bp.blogspot.com/-GaYtSmpZza8/WMx3lx9vGlI/AAAAAAAAAMo/-AkKTbKelYk8IK3Z1vZ-i_GXD2XebfpUQCLcB/s1600/myESP32%2Bpinout.png

mikeriosjr, 2022-07-19:
I get 1 ohm from the c 3030 lid _switch_in pin to:

the the QSB and the xstation board cable plug pin, but at the ESP32 board, nothing at all.

ramapcsx2, 2022-07-19:
Try tracing the signal across the resistor array and the flat flex connectors. Try to find where it ends.

mikeriosjr, 2022-07-19:
New info,

ribbon Pin 20 is showing continuity to the pins between the yellow flagged 3 and 8 on the image attached. 

the lid switch pin shows continuity to its place on the QSB but it doesn't show on the ribbon cable portion

[image]

mikeriosjr, 2022-07-19:
if the QSB has an issue, i can probably directly solder a wire to jump from teal marked point on QSB to the correct pin on the ribbon connector. 
[image]

mikeriosjr, 2022-07-19:
on the xstation board. i only have continuity from 1 side of the R3 resistor array to the ESP32 module.

mikeriosjr, 2022-07-19:
so i think i figured it out. 1 am a dumb dumb. Your yellow marking of 8 is for pin 8 on the ribbon cable board. I'm about 90% sure the QSB board has a broken connection from solder point 8 to pin 8 on the ribbon connection. that is the only spot where there is no continuity. everything is showing correctly

mikeriosjr, 2022-07-19:
All fixed now.  Just soldered a wire from QSB to point 8 on ribbon cable board.

closed 2022-07-19 by mikeriosjr

ramapcsx2, 2022-07-19:
Perfect, that was nice debugging :)
The trace on the QSB might have melted under the iron heat. A wire link is a good fix, it'll work :)