x-station/xstation-issues #236: Vigilante 8 Freeze and Sound issue

RetroGamer90s, 2022-07-23:
Hello 
I love the Xstation. Everything I throw at it works just fine
However, when I tried Vigilante 8 it plays fine on most levels except when playing the level Valley Farms. Sometimes while playing the game would just freeze. Or when so much is happening (cars exploding while interacting with environment) The car engine sound would disappear. 
I am using the latest update for Xstation. released 26/06/2022
I did notice however that the freezing issue disappears when using the update for XStation Firmware + Loader [1.6.1]

The issue happens only on Valley Farms level though. 
Here's a video showing the level 
https://www.youtube.com/watch?v=pSiMMwBmxYs

ramapcsx2, 2022-07-23:
Hey,
can you clarify which xStation firmware produces the freeze, and which works?
"XStation Firmware + Loader [1.6.1] - 2022-06-06" is the latest firmware, it will say "1.6.1" in the loader options.
A bit newer is a loader only update, which doesn't affect the xStation itself.

RetroGamer90s, 2022-07-23:
I really appreciate the fast response!
Ok so initially I was on update "1.4.5" when the freeze happened. Then I only updated to the latest loader from 26/06 and it was fixed. I was still on Xstation version "1.4.5"
later I tried both loader and Xstation version "1.6.1" There is no freeze. I believe only sound issue sometimes but I need more testing. 

However after updating to the loader 26/06 with Xstation "1.6.1" the freeze reappeared again. Sometimes not completing one minute on that level. 
I hope this was not confusing. 

I should also note that Point Blank 2 works in Xstation "1.4.5" with the latest loader 26/06
Thank you

ramapcsx2, 2022-07-23:
Can you try changing your sd card, and test the game again?
When the level runs, all that happens is CDDA playing.

RetroGamer90s, 2022-07-23:
I will try different sd card and report back. 
Right now it is working fine with Xstation "1.4.5" and the latest loader 26/06

ramapcsx2, 2022-07-23:
Well, I'm looking at this game, and that level. There is nothing that happens while the level runs that would be able to cause a hang.
This isn't ruling out a bug, but only saying that a mid-level hang is not something that the xStation would affect.

RetroGamer90s, 2022-07-23:
Ok I tried a different sd card. updated to latest updates Xstation version "1.6.1" with loader 26/06
The freeze happened in that level not 2 minutes in. I always cause the tsunami thing in the level or screw around the water when the freeze happens. Very weird. 
When I play with the old update Xstation "1.4.5" and the latest loader 26/06 the issue is not there.

ramapcsx2, 2022-07-23:
Okay, can you explain to me (a non player of any Vigilante game or the entire genre :p) how to "cause the tsunami thing" please? :p 
Maybe this triggers a CD action, and I can look into that. But from what I'm seeing here, nothing at all happens with the drive while within a level.

ramapcsx2, 2022-07-23:
Update: I pressed some buttons at the end of the waterway, and something in the game changed, and then there was a Tsunami.
This is all the while logs from the CD system were showing: Nothing at all happens then.
If there is a bug, it may have occurred while the level was loading.
It certainly plays fine here for me... I'll keep looking.

In the meantime, do all other games work fine for you?

RetroGamer90s, 2022-07-23:
Sure! :)
Please watch this video around 3:30 minute mark when you destroy any machines on the side You can easily destroy it using the R2 button. Then "tsunami thing" happens lol
Sometimes the freeze happens without it. Sometimes when I try to destroy the machines around the water the freeze occurs
https://www.youtube.com/watch?v=0EwvCSxw-PI  

Yes all other games work just fine I have like 150 games on the sd card tried most of them and they play perfectly 
Again I appreciate your help and support

ramapcsx2, 2022-07-23:
Did you encounter something like this before?:
***Lesser issues***

Vigilante 8
	Has weird red and green(and a few blue) garbage pixels in patches sometimes during gameplay.

RetroGamer90s, 2022-07-24:
Regarding the lesser issuee, there are no issues with other games. I try to find issues or differences berween playing on actual disc and Xstation but it is perfect and only faster loading.
The only issue

RetroGamer90s, 2022-07-24:
Maybe someone else can try the game with latest update and play the level Valley Farms and report back if there are issues like me ?

ramapcsx2, 2022-07-24:
Oh, I copy pasted this little note on Vigilante 8. It had the "lesser issues" text in it :p
Yeah, I'd also like to ask whether anyone else could check out this game..

RetroGamer90s, 2022-07-24:
Update: 
I was testing the level again for about 45 minutes on update "1.4.5" with latest released loader. The freeze happened again.
Then I was confused. So I thought why don't I test the actual game disc on my original Playstation with disc drive, to test if the issue is game related and not the Xstation. So 5 minutes into the level. Bam, it freezes lol
Then I tried it with a PS2 slim and the same thing happened. I kept trying like three times with PS1 and PS2 all froze when I screw around the water and getting hit by the tsunami. 
I play other levels. They play fine. 
I need to note that the CD is actually a bootleg copy and not original one. 

After that I tried it with Duckstation. I couldn't produce the same freeze I tried for almost an hour. Maybe it has something to do with controller vibration ? As I couldn't enable it with duckstaion. Maybe because the emulator works differently then an actual console. I am not sure why ..

So maybe this is a game related bug and not a Xstation issue ??

ramapcsx2, 2022-07-24:
That makes more sense to me at least, seeing as the drive is just idle, playing CDDA :p

RetroGamer90s, 2022-07-25:
Thank you for your help and support throughout. I am glad I own the Xstation :)

ramapcsx2, 2022-07-26:
I'll close this issue, as it's pretty likely some unrelated bug in the game. Should something new come up in combination with xStation, we'll deal with it then :)

closed 2022-07-26 by ramapcsx2