x-station/xstation-issues #237: Discord link?

OmenBoy, 2022-08-05:
The one posted in another thread is no longer valid.

ramapcsx2, 2022-08-05:
These links seem to change behaviour. It's never quite right.. Sorry for that :p
Try: https://discord.gg/ghDefFV4NV

OmenBoy, 2022-08-05:
No problem, thanks :)

icd2k3, 2023-06-19:
Anyone have an updated link? I'd like to check Discord for SD card recs

ramapcsx2, 2023-06-24:
Hey ho,
these :)
[image]

ramapcsx2, 2023-06-24:
https://discord.gg/FTkYwbRG

And a fresh link :)

icd2k3, 2023-06-24:
Thanks @ramapcsx2! I have an ultra and was getting choppy audio and FMV (some games like Castlevania and Parappa virtually unplayable cause of the audio). 

Turns out all it needed was a FULL format via sd card formatter (not quick) and it works perfectly now! Worth mentioning in troubleshooting doc as a potential solve maybe?

swolekat, 2024-01-29:
Looks like the link expired again unfortunately.

ramapcsx2, 2024-01-29:
I found the endless button now xD
https://discord.gg/jQUT7ByfXj