x-station/xstation-issues #238: Ape Escape cutscene stutter & Harmful Park audio stutter

Ghanro, 2022-08-10:
Hi, I'm having an issue with just these two games on a SCPH-5501. I must of tested about thirty other games with no problems. In Ape Escape (USA) on cutscenes only, including the first one before the start game screen, there's a bad audio and cutscene stutter, gameplay is fine. In Harmful Park (Japan) just a bad audio stutter throughout the whole game.

I do have a second SCPH-5501 x-station modded PS1. I've put the SD card into my working PS1 with no problems and then put the SD card from the working one into the other and same issue. Also tried different SD cards. I've tried swapping the cables between the two PS1s, swapping the x-stations, swapping power supplies. Checked the lifted pins for contact, made sure there are is no bridging anywhere. Tested the solder points on the QSB (mmmonkey.co.uk/xstation-sony-playstation-install-notes-and-pinout/) they all make contact. Same issue no matter what combo, one PS1 works fine the other has those issues with just those two games so far that I've tested.

It has the newest firmware. Tried full loading. Any idea why this is happening? Thanks

ramapcsx2, 2022-08-10:
Hey,
thanks for the thorough testing :)

Are there any other modifications installed in these machines? Particularly something like the PS1Digital?

Ghanro, 2022-08-10:
Both have this in game reset mod board (https://www.willsconsolemodifications.co.uk/index.php?route=product/product&path=59_67&product_id=53). It can be disabled by unsoldering the wire going to the motherboard, which I've tried. The only other things I've done is, replaced the capacitors on the power supplies, and swapped the metal housing around the motherboard between the two. The troubled PS1 did have a non-working laser, whereas the working PS1 had a good laser.

ramapcsx2, 2022-08-10:
The IGR should be fine, as well as the other modifications.
Do other games with CDDA show any audio issues? Say, Tomb Raider 1 or 2.

Ghanro, 2022-08-10:
Tomb Raider 1 & 2 work fine. For good measure I tested Tomba 2, Castlevania: SOTN, Jet Moto 2, & Rayman. All CDDA games and they work as they should. Maybe that PS1 just doesn't like Ape Escape & Harmful Park?

ramapcsx2, 2022-08-10:
Hm, all games use one of the few audio methods, and if there was a hardware problem, it should show in others quickly.
I just checked Harmful Park, and that does XA audio. I think Ape Escape is the same.
XA is mostly similar to game data, but it gets sent to the SPU, instead of the CPU...
Anyway, the problem should show up in many other games. Could you try Ridge Racer Type 4 (any ingame race audio)?

Ghanro, 2022-08-11:
Yes, Ridge Racer Type 4 works fine. I've uploaded videos of Ridge Racer and the two trouble games.

https://user-images.githubusercontent.com/110955934/184061243-377094d8-2086-46b6-af00-2115a767dcc1.mp4

https://user-images.githubusercontent.com/110955934/184061278-70352a5c-b3bc-4281-ada7-2da95d0e9d7e.mp4

https://user-images.githubusercontent.com/110955934/184061323-ec371bb4-caa1-4b0f-8427-6d1f4d2dbef5.mp4

ramapcsx2, 2022-08-11:
Okay, this is really weird, and I've not seen such behaviour yet. I'm at a loss for what could be the cause..
At least Ape Escape is clearly also struggling with the video. There could be a sound to video sync in Ape that has to wait while the audio is not working..
But yea, I'm drawing blanks..

ramapcsx2, 2022-08-11:
I've asked around a bit, and it doesn't look like this is in any way a known issue.

A theory is that maybe a timer unit in one of the chips is partly bad, maybe only in some particular timer configuration.
That could explain how most games work, but not these 2.
You should be able to find more broken games, if you look long enough.. but I wouldn't be able to pinpoint the problem either then :/

Ghanro, 2022-08-11:
Bummer, I'll keep testing games. Thanks anyway :)

NDR008, 2022-08-11:
@Ghanro just for completeness sake - did you try swapping the power cable (flipping neutral/live around)?
It is a low likely issue, but some of us got curious "just in case". (The US PSX has a polarized PSU wiring - as far as we know it does not make a difference, but... we are curious).

Ghanro, 2022-08-12:
So I've tried 40 more games and found one more with audio issues. Powerslave. FMVs play fine but in game audio is stuttering like Harmful Park. I tried flipping the power cable and a different cable to no difference. I tried powerslave in my other xstation modded 5501 with the same SD card, and it works fine. Gotta be something about this motherboard that's making this happen. I've uploaded a video...

https://user-images.githubusercontent.com/110955934/184291527-74f72303-ac86-4019-b45f-9ea8fa2bb46e.mp4