x-station/xstation-issues #239: Whole console is half the speed on new X-Station Board, OK with previous one.

Rafo-G2L2, 2022-08-19:
Hi everyone,

I'll try to be as accurate and exhaustive as possible to avoid any ambiguity, so here it is :

I already had an X-Station purchased a few months ago which works perfectly on a SCPH1002.
I also have a SCPH5502 whose drive died, so I purchased a second one, which I received a few days ago.

I installed it with the appropriate board underneath the mainboard and powered it up, and I got a surprise : the whole console is about half the speed, the sound is also half the speed and half the pitch (one octave lower), and after the SCEE logo, I get the "INSERT DISC" view with blinking blocks instead of the playstation screen.
(check video in Google Drive Link below)

First I thought I had done my solders wrong so I double-checked all of them and did not find any issue.

Then I decided to cross-check with my previous X-Station : leaving every else unchanged (the board underneath, the same MicroSD Card with the latest "update.bin" and "loader.bin", the same flat cable), just swap the main X-Station board. Both boards are labeled Rev 1.2a.

And... with the other board everything is running fine : normal speed, I get the PlayStation Screen after the SCEE screen, I get the X-Station menu and I can launch games.

To complete the cross-check I put the "slow board" in the SCPH1002 and.... if gets slow too. which is consistent with a failure on the X-Station Board.

I know that the pin 12 is dedicated to replace the original oscillator, so that X-Station provides her own oscillation for the console so I guessed the frequency that comes out of pin 12 may not be correct, so I checked out : 

The measure between pin 12 and GND for the X-Station that is OK gives a stable 67MHz frequency (see matching video)

The measure between pin 12 and GND on the very same console for the X-Station that is slower gives en erratic measurement (see matching video), which seemed odd to me as I guessed the console should crash with such oscillation jittering, but the video shows a strange signal on the oscilloscope screen : as the wave on screen comes to the right of the screen, it gets "jitterier" (sorry for my inaccurate technical english).

And to add some weirdness to the situation, remember the KO board comes to the "INSERT DISC" on the SCPH5502 ? Well the same board on the SCPH1002 goes to the playstation screen, and then launches the X-station menu, and I can launch games correctly, but STILL AT HALF THE SPEED for display and sound with half the pitch. I could even run a firmware update to the latest from this site : it was slow, but it completed the operation.

### In short : 
OK X-Station on SCPH1002 -> OK
OK X-Station on SCPH5502 -> OK
KO X-Station on SCPH1002 -> everything runs (SCEE screen, PlayStation Screen, menu, games, even FW update) but at half the speed
KO X-Station on SCPH5502 -> SCEE Screen runs, then "INSERT DISC" screen with partially blinking display, also at half the speed.

### Here are my conclusions : 
1. The frequency signal at pin 12 from my new X-Station is not what it is supposed to be, all the rest is working properly.
2. The soldering is not part of the issue, as everything is working perfectly with my previous X-Station board on both consoles.
3. It looks like the SCPH1002 mainboard is more tolerant than the SCPH5502 about oscillator issues, as it does operate correctly, aside the speed issue of course.
4. **I'm not sure it's a hardware issue**, maybe something can be done with a software update (bad parameters when flashing the chip in the factory  ?). If you have some kind of "debug" FW I can try and launch on this board I will be happy to do so and report back too you the results. Getting this board shipped to France was a pain in the a**, so if I can avoid having to ship it back it would be cool.

Here is the link to see all 3 videos : 
- boot on SCPH5502 with KO board
- pin 12 frequency signal on OK board
- pin 12 frequency signal on KO board.

https://drive.google.com/drive/folders/1lut75oJc3Jj_dCjcZazGYFX0afgtuIOx?usp=sharing

I hope I could make myself as clear as possible describing my issue and I hope we can figure out what is going on and solve it.

Don't hesitate if you need me to clarify something.

ramapcsx2, 2022-08-19:
Hey,
that's a broken ESP32 module from what I can tell.
The scope doesn't show it on the waveform, but its frequency reading has dropouts. I assume the clock collapses randomly, then resumes again.

In any case, please contact the seller for a replacement.
This is a new thing. Hadn't had that before! :p

Rafo-G2L2, 2023-01-19:
Some (late) news : the seller sent me a new board, everything is fine now. Thanks for your help.

closed 2023-01-19 by Rafo-G2L2

ramapcsx2, 2023-01-19:
Okay, glad it's resolved :)