x-station/xstation-issues #240: Can't use my memcard pro after install

VardianMX, 2022-08-20:
Hi everybody, i installed a xstation in my 5501 but after the installation my controller port seems like it doesn't give enough energy to my memcard pro or my blueretro adaptor. What is wrong with my playstation?

Thank you all :D

ramapcsx2, 2022-08-20:
Hey,
if the MCPro doesn't come up, maybe the controller port 8V fuse is blown. 
That happens on a different supply rail than the xStation, but it's still somewhat common to happen.
You can test this by using a DualShock pad and see whether rumble works, for example.
If the fuse is blown, it needs to be replaced.

But generally, things that happen on the controller ports are separate from xStation.
If there was a problem with the general 8V supply, then the console wouldn't be able to run anymore at all.

VardianMX, 2022-08-20:
> Hey, if the MCPro doesn't come up, maybe the controller port 8V fuse is blown. That happens on a different supply rail than the xStation, but it's still somewhat common to happen. You can test this [...]

Thank you for your reply :) so i need to replace the fuses PS600 i'll try with another donor console :) thank you so much

closed 2022-08-20 by VardianMX