x-station/xstation-issues #241: Sudden black screen no LEDs on xstation.

thesunnymachine, 2022-08-30:
It seems like my xstation has decided to just stop working.  

It was working fine until now. 

I reflowed the QSB three times and still get nothing on the xstation anymore.  Black screen on power up. 

Can I replace the esp32 chip as I believe it's dead.  

Is it even possible to swap it out?

ramapcsx2, 2022-08-30:
Hey,
if not even the blue LED comes on, that sounds like a power failure.
Was the console running when this happened, or was it a cold start to black screen?
The blue LED is indicating whether any power arrives at all, so not having that could also be a console side issue, for example if the +7.5V supply is broken / a fuse blew / there is a short.
If none of that applies, then the xStation needs replacing. Please contact the seller in that case.

thesunnymachine, 2022-08-30:
> Hey, if not even the blue LED comes on, that sounds like a power failure. Was the console running when this happened, or was it a cold start to black screen? The blue LED is indicating whether any [...]

So I thought there was no signal, as I was testing the system, but I missed the 1 moment it did work.  I had removed the pull up/down resistor off the SD cable and put the console back together enough to test.  I realized I needed the top lid to fully boot, so as I was moving the mount.... What I think happened is the SD card connector has a metal housing.... and it touched the chip somehow causing it to short or something. 

I didn't realize the system had booted because I wasn't looking at my capture card yet, as you can see in this clip below, the video signal goes from the bios to these weird lines then turns off.  You can hear when the SD mount touches the chip in the clip too.  After that the Xstation never showed power again.  

https://youtube.com/clip/Ugkx-c7LKIzHzrHG2Cz4Jf3-L0wU3hhJwypE

Would some Kapton tape protect the chip from this kind of situation in the future?  I barely touched the chip with the SD card mount.

ramapcsx2, 2022-08-30:
Yep, okay. Something shorted out and now there's no power.
An SD card extender will have a metal casing, which is usually grounded.
If this contacts the xStation pcb top side, there aren't many points where it could short out. Most surfaces are covered or ground themselves.
But of course, it probably touched a live part.
With your multimeter, can you probe the voltages going in and out of the xStation voltage regulator?
You want about 7.5V coming in, and 3.3V coming out.

thesunnymachine, 2022-08-30:
Input - 2.34v
Output - 1.35v

I am talking with Castlemainia support and I will link them this thread as well.  I need to avoid this issue in the future since I am planning on modding about 4 other consoles to giveaway.  :/  I could have avoided this had just trusted the resistor removal would work and put the entire console back together.

ramapcsx2, 2022-08-30:
I'll ask around what others make of these voltage readings. To me it sounds like a permanent short on the xStation, yeah.

If Castlemania is swapping your xStation, it'd be great if you could send yours back to them. I'd like to find out what, if anything, shorted so badly.

thesunnymachine, 2022-08-30:
> I'll ask around what others make of these voltage readings. To me it sounds like a permanent short on the xStation, yeah.
> 
> If Castlemania is swapping your xStation, it'd be great if you could send yours back to them. I'd like to find out what, if anything, shorted so badly.

I am still waiting to hear back from them.  Hopefully they will let me do an exchange.   Right now they just want to make sure I've done the basic testing.  It would be great for you to be able to figure this out.  I tested all of the components and nothing seems to be shorted to ground.  

Maybe the chip's programing got reset or something.  I'll ask them to take my old one back so you can figure this out!

ramapcsx2, 2022-08-30:
Update: Could you measure the 7.5V point with the xStation not attached?
You can get to it on the PSX board underside, it's pint "2" here:
https://user-images.githubusercontent.com/6278722/112033652-f08d7380-8b3d-11eb-9124-131df4cb646f.png

ramapcsx2, 2022-08-30:
"Right now they just want to make sure I've done the basic testing"
Common procedure, filters out 90% of user errors, I assume :)
But yeah, if you link to this issue, with a remark saying "certified fried" or so, things will be clear.

I would swap the xStation btw, as long as it's a one time mistake :)

thesunnymachine, 2022-08-30:
> 

Hmm.. 
I am getting 2.45v on that.

ramapcsx2, 2022-08-30:
Alright, this means your xStation could still be fine :)
On the other hand, you now have to find where the power supply 7.5V get dragged down so much.
Maybe a filter burned out and went high ohms.

thesunnymachine, 2022-08-30:
> Alright, this means your xStation could still be fine :) On the other hand, you now have to find where the power supply 7.5V get dragged down so much. Maybe a filter burned out and went high ohms.

Is there a diagram so I can trace the line?  I have tested all the 0 ohm resistors and filters I could find on the board.

ramapcsx2, 2022-08-30:
Already on it as you were typing :p 
https://i.imgur.com/JzVOkGu.png

thesunnymachine, 2022-08-30:
Ok.  I have no idea what I am doing but... I fixed it LOL. 

I couldn't find a few of the listed components labeled on the board.  But it pointed me into the area where the power goes.

I compared the board against a known good board and found a section that was reading differently.

[image]

The circled area in red was where readings were off.  I first replaced the mosfet/Voltage Regulator (sorry I don't know much) looking thing.  And the black filter/cap thing under that.

Then I tested the filters (i think they are filters) with the '20' on them.  The middle one was not putting out 7v like the others.  I removed it and took one off a know good board.. and it worked! 

[image]

[image]

The other board also works without this filter.  

Thanks so much for pointing me in the right direction!  
Do you have any idea where I can get some of these filters to replace the one I took from the working donor board?  Or what I can google to find them?

THANKS AGAIN!

ramapcsx2, 2022-08-31:
Awesome :)
We speculated whether it was the fuse or the filter :p 
So these components with the "20" mark are the fuses. PS601 went high ohms due to the short.
I usually salvage these from dead PSX boards, as I don't need that many.
You want either one such donor fuse, or find an alternative. It should work at 8V and slow blow at 0.8A.. is all I know :p

thesunnymachine, 2022-08-31:
I did some googling and it looks like this might be the fuse. 

US $2.49 | 10pcs Micro Chip Fuse SMD SMT 1210 2A 72V CCP2E CCP2E20 Marking Code 20
https://a.aliexpress.com/_mNgK4dw

The board is also marked with ccp2e20.  Does this seem right?  I'll probably order some and see if it works on the PS1 that I took the fuse from. 

Thanks again for the help!  This was the most complex fix I've had to do.  I am comfortable with soldering and I can follow instructions well, but diagnosis and fault finding is something I haven't done until now with electrical like this.  

I couldn't have done it without your help.  :)

ramapcsx2, 2022-08-31:
These fuses might be it, yeah!
It should work fine, as far as fuses go. They either allow the requested amount of current, or they don't (blow) :p 

Glad this turned out well. Thanks for going the extra mile and working to fix it! :)

closed 2022-09-01 by thesunnymachine