x-station/xstation-issues #242: [Feature suggestion] Favorite list or lists to filter games

FloppySoft, 2022-09-07:
It would be awesome to mark favorites and filter the displayed games accordingly. This would simplify picking your favorite games.
E.g. circle button marking a favorite and SELECT to enable/disable the favorite filter. Furthermore, favorites could be marked in another color or via an icon on the left hand side of each name if the whole game list isn't filtered.

Alternatively, multiple lists could be added (auto incremented list names) and switched via L2 and R2, renaming possible via editing some config files on a PC. This would allow e.g. Todo lists, favorite lists, genre lists etc.

Why would I prefer this to doing this folder-based? Becaue this would work on the fly, adding games to lists while using the PS1. Furthermore, if multiple playlist would exists, a folder based solution would need much storage space.

ramapcsx2, 2022-09-07:
Definitely. It's very high on my todo list, but I'm perpetually in a loop to improve the low level stuff. One day, I'll find that time! :p 

It would be a big improvement to have any kind of "list system". There's then many suggestions already what to do with them.
Mine would be a history of the last played games, to quickly jump back to ;p

infiniteoutrun, 2023-09-26:
Any updates on this feature?

I'd love a favourites list or a 'recently played' list, it would be hugely appreciated.

Libraries can get pretty big (1,000+) these days so it would really help.

ramapcsx2, 2023-09-26:
I'm currently working on xS, the core again though. There's no work yet on the loader with regards to the lists system, but maybe now is a good time :)