x-station/xstation-issues #243: XSTATION.EXE on GameShark for instant boot loops to list when FullBoot chosen

Haker120, 2022-09-13:
I flashed XSTATION.EXE previously pushed into ROM by Docker by some certain programmer that generously did that for me and I cannot believe how FAST I see games' list on screen and this method has one issue, I'm not sure if it's XSTATION.EXE or console's fault but fast boot doesn't work, I mean it does but instead of turning on the game it instantly kicks back to games' list. Fast boot works as intended but could this be fixed/repaired by software or it's console's limitation that only unplugging GameShark will help?

ramapcsx2, 2022-09-13:
Hm, I can take another look at supporting this better, sure. Maybe I can fix it ;)

JonathanDotCel, 2022-09-13:
@ramapcsx2 
There's a magic number which can be written:
*(volatile ulong*)0x80100060 = 0xFFEE2244;

to prevent romprod and unirom from reloading on reboot:
https://github.com/JonathanDotCel/RomProd/blob/171eb3d5282e9f8bf1e92cced57250b1d1fd25fc/rom_shared.s#L82

I don't imagine it would cause issues with games.

Haker120, 2022-09-14:
That would be great! Also I noticed with old xstation.exe boot was way faster than full cycle but old xstation on GS and new on SD slowed down entire boot process, not drasticly but visibly, after update it's again instant. However oddly enough (for me) it didn't kill functionality like ekhm, other similar device. :D

ramapcsx2, 2022-09-14:
There is a series of optimizations to get fast responses in the loader, as well as fast boot times.
When the loader runs off of a ROM cart, all those timings are not fitting the changed environment.

I'll have to optimize for ROM boot, and for that, I first need to know reliably whether it was actually a ROM cart boot.
Ideas, @JonathanDotCel ? :)

JonathanDotCel, 2022-09-14:
Ship it!

Haker120, 2022-09-14:
Well, would be nice to get update for SD and at the same time ROM to flash GS with from main devs and not to ask that generous m8 to make one for me. :D