x-station/xstation-issues #244: Problems with installation

jvivallos, 2022-09-15:
I've haven't been able to use my xstation, last fix was using a new microsd card and it worked just once. Now I have a memcard pro, is there a way to debug problems using unirom?

ramapcsx2, 2022-09-15:
Sure,
Unirom has a status page, please take a picture of that.
Then please describe what it says when trying to boot a game/the loader.

jvivallos, 2022-09-15:
[image]
This is the status, it seems that unirom can't load games from a japanese ps1.

Another question aside from that, the xstation has to be seated on the rf shield to boot right?

ramapcsx2, 2022-09-16:
CD Unlocked says "6"? That is odd. But it doesn't point to anything in particular.
Okay, when you say it has to be on the RF shield, do you mean it doesn't run if not on the shield? (Ie, if it was missing that ground point..).
It will certainly run if only dangling off the flex cable.