x-station/xstation-issues #245: Force PAL games in 60hz?

HasdelBora, 2022-09-22:
Is there a way to make an option for doing it on xStation?

ramapcsx2, 2022-09-23:
Hey, PAL/NTSC patching on the PSX is not easy. Most automatic patches cause issues with the games.
This is best left to old game patches, if they exist, or a dedicated auto patcher, such as Import Player Light.
You can use Import Player Light as the first disk of a disk queue, with your intended game as 2nd. This way you can see how it works out :)

HasdelBora, 2022-09-23:
Thank you for the quick answer :) I tried Import Player Light but screen is not centered (and the centering feature with the pad doesn't work) I'll still go with the PS1 Core on my MisTer so... The 60hz PAL forcing works well :)

L10N37, 2022-10-20:
If you're using a CRT grab a colour correction module for PS1 off ebay. This forces PAL colour frequency permanently Into encoder so you get PAL60.

Otherwise you can use an NTSC GPU frequency Oscillator into the VCKN-A pin for true dual region.

VCNP-A (PAL) is tied to VCKN-A (NTSC) from factory tho so you have to cut trace of pin lift.