x-station/xstation-issues #246: Audio pauses with two different Samsung SD cards

f2bnp, 2022-09-26:
Hey there, about a month ago I modded my SCPH-5502 with this beautiful ODE and I have been very satisfied with it for the most part. I finally have a reliable way to sit down and play my favorite games (and discover "new" ones!) without worrying the laser might die on me at any second.

Unfortunately, I've been having what seem to be reading issues that last a few seconds (~ less than 10secs )that usually manifest as audio pauses. Games affected include: Tekken 3, R4 Ridge Racer, Castlevania SotN, MediEvil
On the above, tested titles, music will sometimes abruptly stop playing for a few seconds and then come back on again. I haven't had any game lock up (I spent many hours beating all four games mentioned), the most severe the issue has been was with SotN, where if I was exiting an area while this issue was in effect, the game would pause for a while, seemingly attempting to load. During such moments, it could take upwards of 30secs for the game to proceed.

I initially thought this was an issue with my Samsung MB-MC128KA/EU EVO PLUS 128GB SD card, so I went ahead and picked up a Samsung MB-MD256KA/EU PRO PLUS 256GB, but the problem persists, though the pause times seem to have decreased ever so slightly, but maybe that's just placebo. I thought I'd be safe with these options, but I guess not. I'm using images strictly from the latest ReDump project and I did use the SD Card Formatter utility to format both cards before putting them to use. Could it be an issue that both cards are close to full (~10% free space) ? Is it these cards themselves that are the issue here? If so, which card should I pick up for the most trouble-free experience, is there a compatibility chart/list ?

Thanks in advance for any help!

maniaco007, 2022-09-26:
Using EXFAT

f2bnp, 2022-09-27:
Could you be a little more specific? Is this a question or a suggestion? One of the cards is 128GB and the other is 256GB, I'd imagine both can only be (and have only been) formatted as EXFAT, no?

f2bnp, 2022-09-28:
Heya, thanks for the reply! I'm kinda glad it's just an SD issue and not something else. Is there a compatibility table or list somewhere?

I've found two possible candidates, the "SDSQUA4-128G-GN6IA" and "SDSQQNR-128G-GN6IA", are both of these suitable or only the latter? Cheers!

ramapcsx2, 2022-09-28:
Pick any standard Sandisk and it'll be fine. The A2 class cards are faster than the A1, but it's not a requirement :)

f2bnp, 2022-09-28:
Many thanks, I'll give it a try!

f2bnp, 2022-10-04:
I ended up going for the slightly more expensive A2 card, so far so good. Avoid Samsung for X-Station, I guess!

closed 2022-10-04 by f2bnp