x-station/xstation-issues #248: Another of the Dreaded "Booting to memory Card"

ReelTechnoFreek, 2022-09-30:
Hello from Ireland.

This is my second xstation install. My first install on a PU18 went very smoothly and worked first time.
My second install on a late PU8 is booting straight to memory card screen.

I have read through many of the issues logged here and this is what I have been able to test/verify so far but still resulting in booting to the memory card screen.

When I power on the playstation I get an immediate green light flash on the xstation. A few second later I get a second green flash and then no more. The blue light stays on all the time.

Once or twice, a track has been recognised in the cd player and pressing play, it plays with no audio. the timer didn't seem to move particularly slow. But 9 out of 10 times no track detected in CD player.

Also checked:

- Correct pins lifted and no bridges
- Correct filter and resisitors removed
- Fluxed and reworked all contact points on QCB
- tested continutity on all points from QCB to mainboard
- lid closed!

Going back over all over issues logged here, the only other thing I can see to try is the lid switch continiuty but not sure how to do that.

Any other guidance would be greatly appreciated.

Thanks in advance

Photos of install to follow.

ReelTechnoFreek, 2022-09-30:
[image]

[image]

[image]

[image]

[image]

[image]

ramapcsx2, 2022-10-02:
Hey,
partly working CD track display means that some signals are only partly making contact. There is some kind of installation issue, most likely.

The lifted pins seem okay and cleaned up underneath. The QSB also looks correctly soldered. I would try reflowing all the connections anyway, see whether the situation improves.

ReelTechnoFreek, 2022-10-03:
I've touched up all the contact points multiple times with no change. Any other thoughts?

As a backup plan I have a 5 series ps1 in the post so will try that with the other QSB I have. Fingers crossed that one works as expected.

ramapcsx2, 2022-10-03:
Well, it's hard to guess what else could be wrong. I go at these issues with an oscilloscope..

ramapcsx2, 2022-10-03:
Best clue so far is that there's occasional track display in the CD player, but most of the time, none.
It's pretty clearly a signal not connecting.