x-station/xstation-issues #249: What causes 2 green blinks then Mem Card / CD Screen?

brendanseattle, 2022-10-01:
Hiya, I successfully installed two xstations at the beginning of the year. I had a bunch of issues even though the installs had no physical problems and had full continuity after testing the points many many times. But I got them to work by reflowing the pins of the chips and deep cleaning.

Sold one, played on the other for several months with relatively no issues in games. Eventually it started to sometimes boot to Mem Card screen, but almost always, if I hit the reset button it would boot to Xstation. Cool.

Last week though, I turned it on and get Mem Card screen. Now I can't get it to boot to Xstation at all. Tried many combinations of cold boots, warm boots, reset, lid open, lid closed, etc. No dice.

I am getting 2 green blinks on powerup, then no further blinks and Mem card screen: https://imgur.com/a/cK9tV8S

What does this code mean? I know when it's a steady blink that the SD card isn't being mounted. But if just two blinks means something, it would be good for me to know for troubleshooting. Either way, I'm expecting to have to open it up and reflow all my solder points and possibly the chip pins again. Just thought it would be good info to have before I start troubleshooting. Thanks!!

ramapcsx2, 2022-10-02:
Oh, it just means that the xStation is trying to send signals, but something doesn't receive them or the response doesn't come back.
So this is some kind of installation / soldering issue. The previous unreliable results also hint at that.
Try reflowing once more, try cleaning the flat flex cable contacts.. Some signal isn't coming through.

brendanseattle, 2022-10-26:
I'm happy to report the issue was a faulty QSB. I had it in my main console, then transplanted it to another after I encountered this issue. Still never got to the XStation screen. 

I even altered the second install to use wires on the throughholes. Nothing worked even though the test points always showed continuity on all the right pins, and nothing was bridging. 

Sure enough, ordered a new QSB from Castlemania and quickly installed it today. First try works perfectly. Glad it was the cheapest part of the hardware.

closed 2022-10-26 by brendanseattle

ramapcsx2, 2022-10-26:
Awesome that it worked!
The QSBs aren't too complicated, but maybe a trace broke, or the connector possibly?

brendanseattle, 2022-10-26:
I have a feeling a trace was broken or corroded from previous flux, but I wasn't able to find anything from visually inspecting / testing continuity. Wouldn't be surprised though as I had installed and reinstalled that QSB half a dozen times or more. I'm glad to just switch over to a new one.

I wanted to use this opportunity to say a MAJOR thank you Rama for all your work here. Not only is the Xstation brilliantly designed and executed, but reading through all these threads and seeing just how much support you provide to anyone willing to ask is inspiring. Seriously don't know how you have the time and energy to be able to help out so many people! Again, a million thank you's. 🙌