x-station/xstation-issues #250: Music audio skipping and breaking up

CMR779, 2022-10-06:
I'm having trouble with music dropping out or skipping during games.  I've heard samsung cards can have this issue.  The card I'm using is a 512GB PNY Pro Elite.  I was sourced from the PNY shop on Amazon.  I'm using split bin cue/bin files, and have the game folders placed in alphabetized folders.  Is there a recommended brand card to use?  If there are any tests I can do to help, let me know.

ramapcsx2, 2022-10-06:
The best test if you suspect the card is to try a different one. In my experience, almost all cards work fine, but there's always one here or there that tends to have slow sectors.
So, try a different card, see whether it helps. You certainly shouldn't get music skipping normally.

Disciple1982, 2023-05-01:
Just wanted to chime in here as this might be helpful to people in the future. I was having this same issue, music was dropping out in the first two games I played, R-Type Delta, and ZanacXZanac. Was using a Samsung SD card with an SD card extender. 

Removed the SD card extender, and replaced the Samsung card with a Sandisk and the music dropping out issue appears to be resolved. I played a couple hours of both games, and did not have any issues with the music. Crossing my fingers and hoping this is the solution but it appears to be the case. :)

CMR779, 2023-05-02:
I switched to Sandisk card as well, and haven't had any issues.  Thanks for commenting.

> Just wanted to chime in here as this might be helpful to people in the future. I was having this same issue, music was dropping out in the first two games I played, R-Type Delta, and ZanacXZanac. [...]