x-station/xstation-issues #251: Install on SCPH-1000 PU-7 1-655-322-11

noromamai, 2022-10-08:
Supposedly the PU-7 is supported, but the IC on the top side has a different number (CXD2516Q instead of CXD2510Q) and the QSB I have (100X QSB v1a) does not seem to fit. Is there any way to install the xStation in this model PU-7?

[image]

[image]

ramapcsx2, 2022-10-08:
Yep, the PU-8 series is fully supported. This starts with the SCPH-1000 consoles that came after the launch in Japan, over to all the international SCPH-100x seriers, and ends with the last PU-8 boards in the SCPH-3500 or so series in Japan.

noromamai, 2022-10-08:
Hi, thank you for taking the time to answer. But the question was about PU-7. PU-7 is said to be supported, but I can't find any install info on it and the QSB does not seem to fit either. 

Checking playstation models wiki page, PU-8 boards on NTSC-J consoles were found in SCPH-3500 and SCPH-5000. If PU-7 is not supported, I will try and get a PU-8 from Japan, but also not sure if that is supported because SCPH-3500 is not listed as supported and SCPH-500x support may exclude NTSC-J consoles, since the SCPH-1000 are all PU-7 and if PU-7 is not supported it may mean the same thing for SPCH-5000.

ramapcsx2, 2022-10-09:
I'm sorry there is confusion.
There is no QSB for easy installation on any PU-7. These boards can only be found in release consoles in Japan, that have the ID of SCPH-1000.
There are SCPH-100x consoles (PAL,NTSC-U) that use PU-8 (EARLY), for which a QSB exists now. These are common to find, rather than the few NTSC-J consoles with a PU-7 board.

If you're looking for a PU-8 board from Japan, then a good chance is on a SCPH-3000 or SCPH-3500. A few SCPH-5000 may also have it, but it would be more likely to be a PU-18 then.
My guess is that SCPH-3000 is likely to have PU-8 (EARLY), and SCPH-3500 and some SCPH-5000 likely to have PU-8 (LATE).

In general, xStation supports (and has a QSB) for PU-8 (EARLY), PU-8 (LATE) and PU-18 boards.
So any Japan console that is not an SCPH-1000 will be supported.

As an aside:
When you read that PU-7 is supported, that meant that the xStation is able to run on these boards, but the the user still would have to install it in one (without the aid of a QSB).
I will be looking into a little guide to do this, as another user also wanted to try this out :)

noromamai, 2022-10-10:
ok. thanks for the clarification. would be interested in seeing a guide for it. in the meantime i will go hunt for a scph-3000/3500

ramapcsx2, 2022-10-10:
I'm working on the next firmware update, which has to pass testing on PU-7 as well. When I'm getting to that, I'll see what I can do for a little guide :)

Tyson1911, 2022-12-31:
Is there anything new in the installation of a PU7 board?  I'm trying to do it myself but it fails to no avail, I'm missing like 4 or 5 points that I can't figure out where they are soldered

ramapcsx2, 2023-01-01:
I'll prepare a little guide for it soon :)

ramapcsx2, 2023-01-03:
It certainly doesn't help that the PU-7 DSP is undocumented and the pinout not known :p
(CXD2516Q, 100 pin)

ramapcsx2, 2023-01-04:
[image]
Does this cover your missing points? Untested, just completed :p

ramapcsx2, 2023-01-04:
[image]

ramapcsx2, 2023-01-04:
DSP 2516Q, lift:
SENSE, C2PO, SUBQ, SCOR
[image]

ramapcsx2, 2023-01-04:
I've got it working :>
The firmware needs some adjustments, but it'll be fine :)

It won't work for you in this state though. It'll require a firmware update.

duncan1111, 2023-01-17:
Iv just installed this on a pu7 - is the firmware available? Thanks