x-station/xstation-issues #254: First Install Trouble - Burning smell, stuck on wh8te sony screen

BakerB31, 2022-10-24:
First time install here.

I confirmed continuity on all points of qsb.

Only trouble i had was lifting first pin and i think the pad lifted as well. Is this a problem?

Screen sits at white sony logo. After a few minutes of it on, screen was fli kering a bit and i smelled a bit of burning.  Any ideas?  Pics attached.

[image]
[image]
[image]
[image]
[image]
[image]

ramapcsx2, 2022-10-24:
Hey,
the installation requires a bit more experience. It's not a good first or second project.
That said, the pictures suggest that at least 2 points are not connected, so it wouldn't have worked.

The smell suggests that something shorted, indeed. What to do next depends on which component died.

Some general tips on soldering:
- The correct iron heat is one where the work goes easy (solder melts nicely), and not more. A too high setting quickly burns the PCB and traces.
- When lifting pins, find a thin and sharp metal tool that can get between the pins nicely. Apply heat to pin and gently move the tool. When done right, the pin gives no resistance to being moved. Once it's free from the pad, gently lift it away and clean up the remaining pad area.
- With your multimeter, check for shorts between lifted pins and their surrounding. Include the pad, too.

I think this board needs to go to an installer, or otherwise be checked out.

BakerB31, 2022-10-24:
I did mean first install of an xstation, I have done some other work in the past, but probably should use lower temp.

What 2 points did you see not being connected?

I confirmed there are no shorts on the chip where pins were lifted just now.

ramapcsx2, 2022-10-24:
Right side of the QSB, nearby the square chip (the Mechacon):
https://user-images.githubusercontent.com/116582766/197575786-9211d096-95eb-4e8c-b9dd-3570410849f7.jpg

ramapcsx2, 2022-10-24:
As I zoom into the mechacon, is there a solder ball on the top row, 5th pin from the left?

BakerB31, 2022-10-24:
> As I zoom into the mechacon, is there a solder ball on the top row, 5th pin from the left?

No, there was jist some flux paste I guess didn't clean up well. I didn't touch that chip.

Also, I mention that i checked continuity on all the qsb pins, so that is not the issue. And i checked for any bridging at all points that i worked on, on the board, and found none.

ramapcsx2, 2022-10-24:
The problem with checking for shorts is that actual shorts regularly get missed, due to various reasons (one being that the pressure of the test probe moves the part).

As your first step now, you need to find which part burned. Try the fuse PS601, which passes the 8V DC supply.