x-station/xstation-issues #255: Pop'n Music Controller Issue on XStation Menu

arobotbitme, 2022-11-07:
Hello!

This is a niche problem, but the way the Konami Pop'n Music controller announces itself to the system sends continuous "left" and "right" inputs to the PS1. The eight or so Pop'n Music PS1 games look for this specific combination of inputs, impossible to do on a normal controller, on boot in order to check whether the Pop'n Music controller is being used so that it can change automatically toggle the corresponding control scheme. Understandably, feeding left+right to the Xstation menu makes it go berserk.

The workaround is booting the system without the Pop'n controller plugged in, selecting the Pop'n game with a standard pad (I keep one plugged into the second port for this purpose), then quickly plugging the Pop'n controller in during the short interval between the Xstation menu and the moment when the selected Pop'n game has booted. This is fine, of course, if a little awkward. **Any chance the Xstation menu can implement Simultaneous Opposing Cardinal Direction (SOCD) cleaning such that an input of left+right registers as no input at all?** This way Pop'n players could boot the system up with a Pop'n controller in the first port, a pad in the second, and use the pad to get where they want without hurrying back to the system to plug the Pop'n controller in before the game boots.

No worries if not. I can't imagine this issue affects too many people or games, though other special peripherals may use a similar recognition method and would benefit from SOCD cleaning.

Thanks!

JonathanDotCel, 2022-11-07:
Hi @arobotbitme, thanks for the detailed description.
Based on what you've said, it should be easy enough to implement this :)

ramapcsx2, 2022-11-07:
Hey ho, as said, this should be doable :)

arobotbitme, 2022-11-07:
Wow, thanks for responding so quickly and being receptive to fixing an extremely esoteric issue! It's awesome to be able to play these on original hardware.

Thanks so much!

ramapcsx2, 2022-11-07:
No promises, but it sounds like the same thing that works for ps1digital hotkeys might work here :)

ramapcsx2, 2022-12-29:
Hey, sorry for the delay, but the next release is in beta testing now, and it has a fix for this included.
If you want to check it out early, it's pinned in the Discord channel: https://discord.gg/pNPA9whA