x-station/xstation-issues #256: Simultaneous .zip file extraction produces loading error

CasualGamerCC, 2022-11-10:
This may just be something to note and not an xstation issue itself. I found that trying to simultaneously extract two or more roms leads to games failing to load and going directly to the CD player screen. If anyone else can reproduce this, maybe it's something to note on the install guide?

For info, I'm using a 16GB Adata SD card, a Surface Pro 3, and installed the xstation into a SCPH-5501 console. Going back and extracting the files in series resolved all the loading issues. I tried once with a single file, which was fixed while the rest still didn't load and then did the remaining ones and they're all good now.

Thanks!

nicolasnoble, 2022-11-10:
You're likely heavily fragmenting the sdcard by doing so. You may be able to "repair" from this state by running the defragmentation tool of your operating system on your card.

CasualGamerCC, 2022-11-10:
> You're likely heavily fragmenting the sdcard by doing so. You may be able to "repair" from this state by running the defragmentation tool of your operating system on your card.

I'm assuming the same thing. Probably just a terrible way to try and save time. I'll use a zip file program that would do a proper sequential decompression rather than the baked in Windows 8.1 thing on my SP3.

ramapcsx2, 2022-11-10:
Okay, thanks for bringing this up. There's just little test data on various SD card issues.
Fragmented files are a big problem for the little microcontroller. It's compounded if the card itself isn't the greatest.
I recommend copying all games in one go, never deleting one and copying another over it (that now fills the free space + fragments).
Another thing is that sometimes, the formatting isn't good for the actual card. A different cluster size might help.
(exFAT is always recommended)