x-station/xstation-issues #257: "Please Insert Playstation CD-ROM" but doesn't seem like a simple PIN 3 issue?

craysiii, 2022-11-12:
I have a Late PU-8 board.

I was having issues with my install so I have already gone and resoldered and tested all points about a month ago and it has been working amazingly this whole time.

However, this morning I went to turn on my PS1 and I was once again presented with the "Please insert Playstation CD-ROM". I took it apart and did all the test points again - they all have continuity. I then tested my PS1 without the top of the console on - it started every time. I then put the console entirely together again, and once again - Please insert Playstation CD-ROM. 

I noticed however, that if I start the PS1 with the lid open to get to the Memory Card / CD Player menu, and then close it after it appears, it boots into x-station every time. If i boot with the lid closed - I get the error message every time.

PIN 3 seems to have a very solid connection, don't have to apply any sort of pressure to see continuity.

Is this still a bad solder issue? Seems very weird..

craysiii, 2022-11-13:
After some discussion with someone (seewood) in the 8BitMods discord, it was recommended that I try a different PSU. I swapped in another PSU from a early PU-8 model I have, and I haven't experienced the issue since.

ramapcsx2, 2022-11-13:
Ah, nice it got solved :)
This might be the reset button being too close to the PSU trigger. I had this issue the other day.

craysiii, 2022-11-13:
Actually, the issue seems to be back now.

I'll probably end up reflowing the thing again. D:

ramapcsx2, 2022-11-13:
Okay, at least this error usually means that one particular signal isn't coming through.
Check around the region of the signal marked as "3":
https://github.com/x-station/xstation-issues#late-pu-8--21-and-later