x-station/xstation-issues #258: Music or sound effects stop working with certain games (Choro Q-2, Castlevania Symphony of the Night re-translation)

Enecororo, 2022-11-13:
Randomly either some of the sound effects, the music, or both would stop, and in Symphony's case the loading for the pause screen will take longer after that happens and the music will sometimes come back as garbage sounds

I can't confirm if the problem goes beyond those two games, but those are the main ones I've experienced issues with

ramapcsx2, 2022-11-13:
Hey, please try a different SD card first. Castlevania is known to show this problem when the card can't keep up.

Enecororo, 2022-11-13:
Noted. But it's going to take some time before I can get my hands on a
different card. What would you recommend?

ramapcsx2, 2022-11-13:
All the Sandisk cards work well, but most other vendors are also fine.
In the meantime, maybe try reformatting the card (exFAT, default cluster size). Sometimes this helps smooth over an unlucky cluster to file mapping.

Enecororo, 2022-11-13:
I can give that a shot. I have a Samsung one

Enecororo, 2022-11-13:
It is a 512GB Pro Plus SDXC

Enecororo, 2022-11-13:
What's an ideal speed/type if I have to get a new one?

ramapcsx2, 2022-11-13:
It's anecdotal, but when there are SD card issues with xStation, it's often a Samsung card.
For speed grades, try to make it an A2 rated one. A1 is also okay though, and both speed grades work perfectly fine.

ramapcsx2, 2022-11-13:
Before you go out shopping though, is there really not a single alternative SD card around?
It is mainly to verify that the issue is indeed caused by the Samsung not playing nice.

Enecororo, 2022-11-13:
I believe the one I have is A2, I'm going to have to double check, once I'm done copying my ROMs over to hard drive. The format is also already ExFAT but I'm going to try reformatting it anyway.

Also the only other SD card I have around is the microSD I'm using for my Memcard Pro.

Enecororo, 2022-11-13:
Does Allocation size effect anything?

ramapcsx2, 2022-11-13:
Please leave it at default, which with exFAT is variable, and chosen based on the total card size.

Why not put an 00xstation folder on the MCPro card, plus your game, and give it a try :)

Enecororo, 2022-11-13:
I'll give that a shot. Worst case scenario if the samsung card is bad for Xstation I can repurpose it for my Steam Deck that's coming soon

Enecororo, 2022-11-13:
the microSD in my Memcard Pro is a 32GB Sandisk Ultra Plus A1 SDHC, should I still give it a shot?

ramapcsx2, 2022-11-13:
Yep, it'll work.

Enecororo, 2022-11-13:
I'll see what happens, if it does work and the Samsung still shows problems I'll drop money on a 512 GB Sandisk once I have money again. I honestly only got the Samsung one cause it was slightly cheaper on Best Buy

ramapcsx2, 2022-11-13:
I'm testing with several cards, and there are some Samsung in it, out of which one older model has issues and the others are fine.
I'm not sure why these come up regularly, but apparently there are Samsung being sold right now that don't hold up with xStation.

Maybe I should get more Samsungs, until I have one that shows the problem..

Enecororo, 2022-11-13:
Yeah I bought mine pretty recently, so it's weird that it's having these kinds of problems, but I'll work it out one way or another

Enecororo, 2022-11-14:
Progress report, Choro-Q 2 still has the problem of the engine sounds dying if you collide into enough things. Symphony of the Night has no problems so far, but I can't really do that on command anyway, So it could pop up any time

Enecororo, 2022-11-14:
Yeah the main issue with Choro-Q is that the engine sounds will stop after enough collisions, though the music also stops in that one too sometimes.

As of the inverted castle the vanishing music has yet to happen using the Sandisk 32GB

Enecororo, 2022-11-14:
The same thing also happens with Choro-Q 3, it's entirely possible it could be game related rather than card related

ramapcsx2, 2022-11-14:
I tested Choro, that looks like a plain game engine bug to me :p
Can't be sure, of course, but there is nothing happening on the CD side that stands out. It's just a bunch of sounds too many, maybe occupying all the SPU voice slots and not freeing them properly.

Enecororo, 2022-11-14:
I can see how it might be a game engine issue. Just you can never be too
sure

Enecororo, 2022-11-14:
No music issues yet but the game did crash right when I tried to save it one time. not sure if that's related to the A1 sandisk or not

ramapcsx2, 2022-11-16:
Saving a game is also rather unlikely to be CD related, though this must've been frustrating for sure ><

ramapcsx2, 2022-11-30:
Hey, any update on the Castlevania music? I think it should be perfectly fine with any well working SD card now.
I'm checking this game with the development firmware, and it really just needs the data it wants in time.

Enecororo, 2022-12-10:
Okay so with my brief testing with the ultra microSD no issues with the music but it did crash during saving

I have an Extreme Pro standard SD card now with up to 200 MB/s read speed, and the music still has no problems, but the crashing during saving is still happening (It could be cause of the quality romhack I'm using)

ramapcsx2, 2022-12-11:
Okay, maybe check Google for any reported problems saving. It's not an ODE issue, or at least that's unlikely :p

Enecororo, 2022-12-11:
I'll just assume it's a result of the quality hack for now, maybe try the retranslation of the Japanese version.

I did also test Choro-Q 3 and there was no issues with the sound effects or music using the Extreme Pro card

Enecororo, 2022-12-11:
After playing the fan re-translation of Symphony up to the Inverted Castle I feel I can reasonably assume that the quality romhack of the English version might have been contributing to the crashing during saving. No music issues either

closed 2022-12-29 by Enecororo