x-station/xstation-issues #260: Issue with Multidisk Games

AVTechMan, 2022-11-21:
Hello, first time posting here as I'm not well versed with so many different fourms and support sites. 

Anyway, I recently did an install of the X-Station on a 5501 console with the PU18 motherboard. The install went fine, and system boots up to the loader and my games are listed. Tested a few games (not multi) and loads and runs like it should.

With that said, I am having trouble with the multidisc games. Having tried two different games (FF7 and 9), I select the discs for the playlist in the menu and upon starting the game (I have a save file for each that requests a different disc) and opening the save that requires the different disc, no matter how many times I open and close the lid, the disc number does not change. I know the lid switch does work properly as I can access the CD app with the lid open when powered on, and once closed it launches into the XStation menu.

Has anyone else had any issues with the disc swap on the XStation and anything else I could try/look at? If nothing else, I suppose I can just reboot the system and select the disc manually but would defeat the purpose of swapping discs in-game. 

I read of a past issue with another person that had the same issue but had a different motherboard revision. I haven't found really many posts regarding the multidisc feature or issues on google or otherwise. Any help would be great!

ramapcsx2, 2022-11-21:
Hey,
first, please make sure you're on the latest firmware, which is 1.6.1 currently.
Next, please create a queue as you like, then power down the console and check the SD card on your computer.
The file "diskqueue.txt" in /00xstation should list all the files in your queue.

If that is correct, please try starting your queue in "full boot" mode, so the BIOS startup runs. Open the lid then, so you can select the CD player.
Once in the CD player, close the lid and watch what CD gets detected.
Each time you cycle the lid now, the detected CD should slightly change (in length, or number of tracks, depending).

If that checks out, everything should be working correctly. Try your game again :)

AVTechMan, 2022-11-21:
> Hey, first, please make sure you're on the latest firmware, which is 1.6.1 currently. Next, please create a queue as you like, then power down the console and check the SD card on your computer. The [...]

Hello sir! 

Firmware is at the latest 1.6.1.

Created the queue and checked the 'diskqueue' file on my PC....all three files were there (FF7 in this case).

Then from the menu booted the queue in 'fullboot' mode with the cover into the CD player app.

When I closed the lid, the time came up as '1:70:39' with one track listed. However when I opened and closed the lid a second time the same exact time appeared with the single track. Third time did the same thing, so apparently the same exact track length and track number appear each time I cycle the lid. 

I'm not sure what the deal is there.

ramapcsx2, 2022-11-21:
It might be that the xStation can't read the lid sensor, a connection error.
The signal in question is marked "8" here: https://github.com/x-station/xstation-issues#pu-18

AVTechMan, 2022-11-21:
Okay I will check this when I get off work today.

AVTechMan, 2022-11-22:
> You can do a test to check this more easily:
> In the CD player, open the lid and take something like a Q-Tip to push down the limit switch.
> Each time you push it down, the green xStation LED should come up strongly. This signifies that it's mounting the new disk.
> The LED should come on the moment you remove the Q-Tip / "open the lid".
> If it does that, it should change the displayed track lengths..

Okay I got to test the LED light in the CD player. No green LED when I pressed the lid switch down so possibly suspect the #8 point in the photo you linked in your previous post so I will inspect that part and see whats up.

AVTechMan, 2022-11-22:
Okay think I finally reached the conclusion. Turns out that though the solder joint was fine on #8 point it looks to me that the chip above it is not communicating with the lid sensor somehow. Everything else works however other than that. 

So rather than going through the trouble of desoldering the QSB I had an extra QSB on hand and several PU-18 motherboards so I soldered the extra QSB to one of them. Transferring the Xstation to the other system the disc swap works now, in the CD screen and the actual game.

So rather than toss the previous system I took the station out of ill just keep it as an extra spare since it still works great with my games including the multidisc sans the swap function. Ill just have to get another Xstation to replace that one soon when I have more funds available. 

Thanks for you help Rama it was very appreciated I can go ahead and close this ticket out now. :-)

AVTechMan, 2022-11-22:
> I hope it's a simple fix like that :)
> But a note: If it works, the LED will come on the moment you open the lid / remove the Q-Tip, not when you close it.

Your right and thats exactly what it did in the replacement console. Soon as I removed the spring during the test the green LED lit up indicating that the swap worked and once the lid was closed after a moment the led was once again reading the data. The former system never did that.

closed 2022-11-22 by AVTechMan

ramapcsx2, 2022-11-22:
Awesome it's fixed :)

I'm sure the other console can be patched up also. It depends on what exactly is wrong, but maybe just a patch wire to that pin will do it :)

AVTechMan, 2022-11-23:
> Awesome it's fixed :)
> 
> I'm sure the other console can be patched up also. It depends on what exactly is wrong, but maybe just a patch wire to that pin will do it :)

Yeah I am sure it can be fixed, but will save it for another day. :)