x-station/xstation-issues #261: Xstation not booting

Killerbee256, 2022-11-27:
I'm having an issue with the xstation I installed this morning. I can't get to the xstation menu. I don't know if it's a sd card or an installation issue. As I write this I'm formatting another sd card to test with it. When the playstation is booting I see the access light on the xstation come on for a moment. Here's photos of my solder work.
[image]
[image]
[image]
[image]

jamesrpatterson, 2022-11-27:
I'd recommend using a continuity tester to go over your soldering of the QSB and make sure each point is making contact.

You can find the images showing the test points here: https://support.castlemaniagames.com/support/solutions/articles/44002135615-test-points-for-pu-18

Killerbee256, 2022-11-27:
I'll give it a try. I'm new at this, thus not very good at it. I tried to re-flow the solder a few times. I hope I didn't damage or mangle anything too badly.

ramapcsx2, 2022-11-28:
Hey,
I recommend you have a look around in other issues like these. 
You'll find the things people recommend for this general type of error.

It's most likely an issue with the QSB soldering, but do the CD player test and see whether a disk is detected there, and then whether it plays normally when you press start.

Killerbee256, 2022-11-28:
I checked the QSB and my soldering checked out. I did the CD player check and yes it detected a "CD" it "played" but there was no sound.

ramapcsx2, 2022-11-28:
Do you have a way to run homebrew? To be able to run Unirom, and get its logs when it tries to read data..
Ways to do this: an expansion ROM (cheat) cart, or something called Tonyhax on a Memory Card.

Killerbee256, 2022-11-28:
I bought a memory card pro to use with the xstation. I will see if I can figure how to run Tonyhax from it.

ramapcsx2, 2022-11-29:
Okay, you just need something to boot this homebrew:
https://psx0.files.wordpress.com/2022/04/unirom_cheapcart_1.png

Killerbee256, 2022-11-29:
Both of my sd cards are from SanDisk, Should I try a different brand? If so what brands of sd cards have people had luck with?

ramapcsx2, 2022-11-29:
Sandisk is fine and those are the recommended ones.

ramapcsx2, 2022-11-29:
If you can't find any issues, maybe consider having the console checked. 
There are some soldering issues that are difficult to find.

Killerbee256, 2022-11-29:
I'll double check the soldering. this time checking for _improper_ connections. There might be a solder bridge I missed. 
Who would have I check out the console?

Killerbee256, 2022-11-30:
I took the PlayStation into repair shop think there's something odd with my Playstation. I ordered another 5501 from eBay.

ramapcsx2, 2022-11-30:
Hm, a general repair shop is fine for things like reflowing, but a more specialized retro modder in you area would've been my choice.
Still, a good reflow should have fixed the more common problems, so maybe the console was indeed defect.

You have another PU-8 QSB, right? Maybe try a cheap SCPH-1001 (in the US?) :)

Killerbee256, 2022-11-30:
I reached out to a few xstation installers in the US but none replied. SCPH-1001 seem to be rare. I called the local retro used game shop but all they had was SCPH-7000s and higher, plus they wanted 59 dollars! So I bought on 5501 like I alright have from ebay for half that price.

ramapcsx2, 2022-12-01:
Okay, makes sense then. Hope this one works out! :)

Killerbee256, 2022-12-14:
The shop installed the xstation in the second PlayStation, and got the same
result. Is it possible that part of the xstation is defective? Between the
shop I've tired a two different SanDisk micro sd cards and one Kingston
branded one. The xstation creates a empty config file so the cards are
being read but the data isn't getting to PlayStation correctly.

ramapcsx2, 2022-12-15:
Hey, that's a major bummer to hear. Yes, of course there can always be a problem with any device. The task is to find out what's wrong.
Where are you located?

ramapcsx2, 2022-12-16:
So I wanted to know that, because if you're in the EU, I'd like to take a look at the console + xStation myself :p
If you're in the US, I'd like to try and match you with a retro modder with whom I've worked before.

But if you want, you could try returning the xStation to the seller for a replacement. It's worth a try, and maybe the xStation is defective (it's rare).

Killerbee256, 2022-12-16:
I contacted castlemania Games, they're going to send a new xstation. Then
I'll send the old one back to them. A retro modder did respond to my email.
But I'm waiting to if the shop I have the local repair shop install the new
one when it arrives.

ramapcsx2, 2022-12-16:
Alright, really hope it gets sorted quickly. I know it's frustrating.

Killerbee256, 2022-12-22:
It was indeed defective, the replacement xstation worked right away. Thank for for you help.

ramapcsx2, 2022-12-22:
Oh, okay! Well, it would be awesome if you could send the defective one back, so I can take a look at what went wrong.
This is rare, but does happen, and I want to learn from it :)

But in any case, glad it's sorted now. Hope you enjoy! :)

Killerbee256, 2022-12-23:
I've got it ready to mail. I'll mail it when the post office opens again.